Xilinx/Vitis-AI #104: Server hang when running adas-detection application in docker container

surekha1723, 2020-05-20:
I am trying to run adas-detection application in a docker conatiner on alveo_u200 fpga card.
But once i start running **source run.sh** command it is stopping at SPEAKING TO BUTLER and if i check my fpga server, it is in hang state and reboots itself.

Expected:
Applicaion should run without hang

Please suggest me how can i avoid his hang

Thanks in advance

sumitn-xilinx, 2020-05-20:
I hope you have installed XBUTLER using the install.sh script mentioned in the "Setup for Alveo" section on https://github.com/Xilinx/Vitis-AI/tree/master/alveo page.
You can try following inside conda environment:
 xbutler_backdoor.sh -s
If the output of above seems good, can you try running following test:
https://github.com/Xilinx/Vitis-AI/tree/master/alveo/examples/deployment_modes

surekha1723, 2020-05-21:
@sumitn-xilinx Thanks for your reply

I Installed xbutler using install.sh but it is getting stopped at Starting butler at 67% ,its not exiting gracefully.I manually stopped and ran xbutler_backdoor.sh and found out examples are working
as expected
[image]

Here is my output of xbutler_backdoor.sh

(vitis-ai-caffe) root@vitispod:~# xbutler_backdoor.sh -s
-------------------------------
Validating Arguments... passed!
-------------------------------

----------------------
Verifying CONDA_PREFIX
----------------------
----------------------
Verifying XILINX_XRT
----------------------
Pass!
Password Correct!
-------------------------
Server System Information
-------------------------

        FPGA #0
                FPGA Name: alveo-u200
                DSA Name: xilinx_u200_xdma_201830_2
                XCLBIN Path: /opt/xilinx/overlaybins/xdnnv3/xdnn_v3_96x16_2pe_8b_9mb_bank03_2.xclbin
                Total Number of CUs: 2
                Total Number of Used CUs: 0
                Total Number of Unused CUs: 2
                CU List (Properties and Status):
                        0)   CU Name: kernelSxdnn_0   Kernel Name: kernelSxdnn_0   Base Address: 0x1c00000   Used: false
                        1)   CU Name: kernelSxdnn_1   Kernel Name: kernelSxdnn_0   Base Address: 0x1e00000   Used: false

**Now my query is if i give /opt/xilinx/xrt/bin/xbutil query -d 0 while running the ai example, the fpga server is hanging.**

sumitn-xilinx, 2020-05-21:
Most likely the issue with your XRT installation.
Can you manually go through each installation step in install.sh?
Which linux distribution do you have?

surekha1723, 2020-05-21:
@sumitn-xilinx Thank you for the reply 

Please find output of lsb_release in my fpga server
 lsb_release -a
Distributor ID: Ubuntu
Description:    Ubuntu 18.04.4 LTS
Release:        18.04

I installed 2.5 verision of XRT and deployment software as well in my server and able to see device info using **xbutil query -d 0** command.
But when i start running this vitis applications in container i am facing the hang issue.

VishalX, 2020-05-22:
@surekha1723 
Did any Alveo example work for you or only adas app is behaving unexpectedly?

Also, can you check if the card is working fine?
You can run (in BASH):
```sh
source /opt/xilinx/xrt/setup.sh
xbutil validate -a
```

surekha1723, 2020-05-22:
@VishalX Thanks for your reply

Not only adas even video_analysis application is also giving unexpected behaviour
Please find below output for xbutil validate command
[image]

Thanks in advance

hanxue, 2020-08-03:
Hi @surekha1723 , seems the last output shows `successful`. Are you still facing problem with XRT?

I have also answered in #105 on uninstalling and reinstalling XRT

qianglin-xlnx, 2022-02-16:
Hi @surekha1723 
Is this issue solved? If yes, could you please close this issue. Thank you.

qianglin-xlnx, 2022-04-11:
Hi @surekha1723 
This issue has been opened for a long time and I'm going to close it. If you still have other questions, please open a new issue for it. Thank you.

closed 2022-04-11 by qianglin-xlnx