Xilinx/Vitis-AI #105: Running XBUTIL hangs server while running vitis ai applications on u200 alveo card

surekha1723, 2020-05-21:
I ran vitis applications in a container ex. adas_detection and tried to see utilisation of fpga using command - **xbutil query -d 0**.
When i issue this command the server is hanging and once it reboots it is saying Card index 0

Please find attached screenshot
[image]

Please suggest me to avoid this issue.
Thanks in advance

hanxue, 2020-08-03:
Hi @surekha1723 , this article could be the answer to your problem https://www.xilinx.com/support/answers/73055.html 

Install the Python 2 `python-pyopencl` package. 

If that does not work, try to reload the XRT kernel modules again

```
$sudo rmmod xocl
$sudo rmmod xclmgmt
$sudo modprobe xocl
$sudo modprobe xclmgmt
$xbutil validate
```

And finally you can try uninstalling and re-installing XRT. 

Are you still facing the problem with XRT?

qianglin-xlnx, 2022-02-16:
Hi @surekha1723 
Since we haven't received your reply for a long time, we assume you have solved this issue and I'm going to close it. If you still have any questions, please feel free to reopen it. Thank you very much.

closed 2022-02-16 by qianglin-xlnx