Xilinx/Vitis-AI #106: DPU-TRD build failure

spaceotter, 2020-05-21:
I built the [official platform](https://github.com/Xilinx/Vitis_Embedded_Platform_Source/tree/master/Xilinx_Official_Platforms/zcu104_dpu) from source, which seemed to work, and the image boots.
After running:
```
 2012  cd ~/src/Vitis-AI/
 2014  cd DPU-TRD/
 2024  export TRD_HOME=$(pwd)
 2027  cd prj/Vitis/
 2028  export SDX_PLATFORM=~/src/Vitis_Embedded_Platform_Source/Xilinx_Official_Platforms/zcu104_dpu/platform_repo/zcu104_dpu/export/zcu104_dpu/zcu104_dpu.xpfm 
 2029  make KERNEL=DPU_SM DEVICE=zcu104
```
I get the following error:
```
===>The following messages were generated while processing src/Vitis-AI/DPU-TRD/prj/Vitis/binary_container_1/link/vivado/vpl/prj/prj.runs/impl_1 :
ERROR: [VPL UTLZ-1] Resource utilization: RAMB18 and RAMB36/FIFO over-utilized in Top Level Design (This design requires more RAMB18 and RAMB36/FIFO cells than are available in the target device. This design requires 1054 of such cell types but only 624 compatible sites are available in the target device. Please analyze your synthesis results and constraints to ensure the design is mapped to Xilinx primitives as expected. If so, please consider targeting a larger device.)
ERROR: [VPL UTLZ-1] Resource utilization: RAMB36/FIFO over-utilized in Top Level Design (This design requires more RAMB36/FIFO cells than are available in the target device. This design requires 492 of such cell types but only 312 compatible sites are available in the target device. Please analyze your synthesis results and constraints to ensure the design is mapped to Xilinx primitives as expected. If so, please consider targeting a larger device.)
ERROR: [VPL UTLZ-1] Resource utilization: RAMB36E2 over-utilized in Top Level Design (This design requires more RAMB36E2 cells than are available in the target device. This design requires 492 of such cell types but only 312 compatible sites are available in the target device. Please analyze your synthesis results and constraints to ensure the design is mapped to Xilinx primitives as expected. If so, please consider targeting a larger device.)
ERROR: [VPL 4-23] Error(s) found during DRC. Placer not run.
WARNING: [VPL 60-732] Link warning: No monitor points found for BD automation.
ERROR: [VPL 60-704] Integration error, problem implementing dynamic region, place_design ERROR, please look at the run log file 'src/Vitis-AI/DPU-TRD/prj/Vitis/binary_container_1/link/vivado/vpl/prj/prj.runs/impl_1/runme.log' for more information
ERROR: [VPL 60-1328] Vpl run 'vpl' failed
ERROR: [VPL 60-806] Failed to finish platform linker
INFO: [v++ 60-1442] [14:58:33] Run run_link: Step vpl: Failed
Time (s): cpu = 00:00:12 ; elapsed = 00:27:36 . Memory (MB): peak = 677.277 ; gain = 0.000 ; free physical = 10257 ; free virtual = 81449
ERROR: [v++ 60-661] v++ link run 'run_link' failed
ERROR: [v++ 60-626] Kernel link failed to complete
ERROR: [v++ 60-703] Failed to finish linking
Makefile:69: recipe for target 'binary_container_1/dpu.xclbin' failed
make: *** [binary_container_1/dpu.xclbin] Error 1

```

wilderfield, 2020-05-21:
@spaceotter It is not crystal clear, but did you follow the steps mentioned here: https://github.com/Xilinx/Vitis-AI/blob/master/DPU-TRD/prj/Vitis/README.md#552-configue-the-zcu104-released-project

You need to specify a few verilog defines to control the generation of the DPU IP.

Hopefully this is the problem.

spaceotter, 2020-05-22:
I guess the defaults don't work for zcu104.
Setting `URAM_ENABLE` appears to have helped, but:
```
===>The following messages were generated while  Compiling (bitstream) accelerator binary: dpu Log file: /home/eric/src/Vitis-AI/DPU-TRD/prj/Vitis/binary_container_1/link/vivado/vpl/prj/prj.runs/impl_1/runme.log  :
ERROR: [VPL-4] design did not meet timing - hold violation
WARNING: [VPL 60-732] Link warning: No monitor points found for BD automation.
ERROR: [VPL 60-704] Integration error, problem implementing dynamic region, route_design ERROR, please look at the run log file '/home/eric/src/Vitis-AI/DPU-TRD/prj/Vitis/binary_container_1/link/vivado/vpl/prj/prj.runs/impl_1/runme.log' for more information
ERROR: [VPL 60-1328] Vpl run 'vpl' failed
ERROR: [VPL 60-806] Failed to finish platform linker
INFO: [v++ 60-1442] [18:28:09] Run run_link: Step vpl: Failed
Time (s): cpu = 00:01:44 ; elapsed = 03:02:59 . Memory (MB): peak = 677.277 ; gain = 0.000 ; free physical = 4809 ; free virtual = 80741
ERROR: [v++ 60-661] v++ link run 'run_link' failed
ERROR: [v++ 60-626] Kernel link failed to complete
ERROR: [v++ 60-703] Failed to finish linking
Makefile:69: recipe for target 'binary_container_1/dpu.xclbin' failed
make: *** [binary_container_1/dpu.xclbin] Error 1
```

wilderfield, 2020-05-22:
You are correct. The defaults are for ZCU102_DPU platform.

I’d expect this to work with the correct modifications. Will check tomorrow.

qianglin-xlnx, 2020-05-22:
@spaceotter It seems that it still use the zcu102 config. could you please check the following.  

[image]

jimheaton, 2020-05-22:
> CAUTION: This message has originated from an External Source. Please use proper judgment and caution when opening attachments, clicking links, or responding to this email.

The following flow works for the ZCU104:
In Makefile.
Change XOCC_OPTs to the following:
```XOCC_OPTS = -t ${TARGET} --platform ${SDX_PLATFORM} --save-temps --config ${DIR_PRJ}/config_file/prj_config_104_2dpu --xp param:compiler.userPostSysLinkTcl=${DIR_PRJ}/strip_interconnects.tcl ```

 Make the follwing edits to DPU/TRD/prj/dpu/dpu_conf.vh
* Change the line with ``` \`define URAM_DISABLE to \`define URAM_Enable```
* Change the line with ``` \`define RAM_USAGE_LOW to `define RAM_USAGE_High```

make KERNEL=DPU DEVICE=zcu104
Note this will not enable the softmax kernel in the PL.

[image]<https://user-images.githubusercontent.com/58016351/82638994-6bde7a00-9c3a-11ea-9822-6d3a12cc2793.png>

spaceotter, 2020-05-23:
The build completed with the suggested changes.
Thanks!

closed 2020-05-23 by spaceotter