Xilinx/Vitis-AI #107: Using DNNDK with custom build models

pentapatisivani, 2020-06-07:
What is the required format if we want to quantize and compile custom models on board?
Each model has a make file. How can we create it for other models? and it has separate directories for different boards. How can we write custom code based on the board we have?
or Can we only run the models present in this link?
https://github.com/Xilinx/Vitis-AI/tree/master/VART/samples
Simply put, How can I deploy custom models on DNNDK?

wilderfield, 2020-06-07:
The samples directory you are examining provides prequantized and
precompiled models that have been prepared for the two default
configurations of the DPU-TRD. (ZCU102 and ZCU104).

To run a custom model on a custom board.

You need to use the latest vitis-ai docker container.

First you need to quantize it using the vai_q_* utility.

You need to generate a dcf file which will have metadata about the DPU
configuration. This file is passed to compiler via arch.json.

Then you need to compile using the vai_c_* utility.

The output of the compiler is passed to the runtimes as you see in the
“samples”.

This is the process from a high level.

I hope this is helpful.

parth1595, 2020-09-23:
Is there any example for any custom model for converting from .pb Tensorflow model file to DNNDK output format file.

andyluo7, 2020-09-23:
https://github.com/Xilinx/Vitis-AI-Tutorials/tree/Keras-GoogleNet-ResNet

qianglin-xlnx, 2022-02-16:
Hi @pentapatisivani 

Since we haven't received your reply for a long time, we assume you have solved this issue and I'm going to close it. If you still have any questions, please feel free to reopen it. Thank you very much.

closed 2022-02-16 by qianglin-xlnx