Xilinx/Vitis-AI #109: Question : Does Vitis-AI support Alveo-U25 Flow ?

ajithAI, 2020-06-18:
Does this Vitis-AI package support Alveo U25 ? 
Does "xilinx_model_zoo" contains U25 support ? 
If so, where can I download U25 xclbin !!

qianglin-xlnx, 2020-07-14:
Hi @ajithAI 
So far as I know, Alveo U25 is not supported by Vitis-AI.

ajithAI, 2020-07-14:
Hi @qianglin-xlnx 
Thanks for your Reply. 
Just, want to know, if any new device has to be supported by Vitis-AI, what all the necessary builds that required. 
I know that, we need XRT, DSA from device side. What all the other modifications that we need to do at Vitis-AI end to support device ? 

Thanks,
Ajith.

wilderfield, 2020-07-14:
@ajithAI the "xclbin" needs to be generated for U25. The xclbin is the binary to be loaded to the FPGA, and it has meta data (like memory addresses) for the host side software. The xclbin is the bridge. At the moment, we don't enable customers to generate the xclbin with our accelerator included for Alveo cards.

sadicLiu, 2020-07-14:
Hi @qianglin-xlnx, @wilderfield , is Zynq-7000 platform supported? I want to run yolo on zynq-7045, and I found the document [here](https://github.com/Xilinx/Vitis-AI#programming-with-vitis-ai) clarified that zynq-7000 is support, but in [here](https://github.com/Xilinx/Vitis-AI/blob/master/doc/faq.md#which-vitis-platforms-will-vitis-ai-support) and [here](https://www.xilinx.com/support/documentation/sw_manuals/vitis_ai/1_2/ug1414-vitis-ai.pdf), it seems that zynq-7000 is not in the support list, so can you guys clarify this issue here? Thanks!

ajithAI, 2020-07-14:
@wilderfield 
Thanks for your reply. 
The Compute Shell ( XCLBIN ), Xilinx might be providing. ( Not yet, May be in near future .. hopefully )
Customers may buy this IP, or subscribe to get the package. 
Beyond the Compute Shell, what all the software requirements that needed to run Examples in Vitis-AI, targeting U25. 
Does VART ( Vitis-AI Run Time ) needs any patches to support U25 ? 

So, What is basically needed to run all the examples ( inside Vitis-ai, eg : ResNet50 ), for any new device, given we have XRT, DSA & Compute Shell.  

Thanks.

qianglin-xlnx, 2022-02-16:
Hi @ajithAI 
Since this issue hasn't been updated in a long time, I'm going to close it. If you still have any new questions, please open a new issue for it. Thank you very much.

closed 2022-02-16 by qianglin-xlnx