Xilinx/Vitis-AI #110: Broken Links

deke997, 2020-06-28:
In the documentation for tensorflow and caffe in /alveo/examples/, there is a link that points to:
https://github.com/Xilinx/Vitis-AI/blob/master/alveo/examples/tensorflow/Benchmark_README.md
and
https://github.com/Xilinx/Vitis-AI/blob/master/alveo/examples/caffe/Benchmark_README.md
respectively
It says "For published results, and details on running a full accuracy evaluation, please see these Instructions", but both links are dead

Where can we find details on on running a full accuracy evaluation?

sumitn-xilinx, 2020-06-28:
Thanks for pointing out. This is about to be updated very soon.

Which models are you trying to benchmark?

There are some caffe examples (googlenet, resnet50, yolov3 etc.) in alveo/apps/aks. These can be modified to run tensorflow models as well. More examples are going to be added early next month.

deke997, 2020-07-02:
U200

hanxue, 2020-08-03:
Hey @deke997 , do you still find broken links in the latest v1.2 release? I look at the following directories 

https://github.com/Xilinx/Vitis-AI/tree/master/alveo/examples 
https://github.com/Xilinx/Vitis-AI/tree/master/alveo/examples/caffe
https://github.com/Xilinx/Vitis-AI/tree/master/alveo/examples/tensorflow

and there is no longer a link to the missing https://github.com/Xilinx/Vitis-AI/blob/master/alveo/examples/tensorflow/Benchmark_README.md 

I am curious about your resnet50 caffe benchmark results. Would you mind sharing your results here, and what kind of results you see?

closed 2021-01-04 by qianglin-xlnx