Xilinx/Vitis-AI #113: ZCU102 board booting Error

bettertm, 2020-07-08:
follow "https://github.com/Xilinx/Vitis-AI/tree/master/VART#running-vitis-ai-examples-for-edge".  I Flashing the OS Image to the SD Card. But There is something wrong, when I boot the evaluation board using SD card.  Please help me!

Starting kernel ...                                                             
                                                                                
[    4.474508] Kernel panic - not syncing: VFS: Unable to mount root fs on unkn)
[    4.482933] CPU: 1 PID: 1 Comm: swapper/0 Not tainted 5.4.0-xilinx-v2020.1 #1
[    4.490056] Hardware name: ZynqMP ZCU102 Rev1.0 (DT)                         
[    4.495004] Call trace:                                                      
[    4.497441]  dump_backtrace+0x0/0x148                                        
[    4.501090]  show_stack+0x24/0x30                                            
[    4.504390]  dump_stack+0xac/0xd0                                            
[    4.507696]  panic+0x148/0x314                                               
[    4.510734]  mount_block_root+0x1d4/0x288                                    
[    4.514725]  mount_root+0x128/0x15c                                          
[    4.518198]  prepare_namespace+0x164/0x1ac                                   
[    4.522277]  kernel_init_freeable+0x250/0x274                                
[    4.526620]  kernel_init+0x18/0x104                                          
[    4.530099]  ret_from_fork+0x10/0x1c                                         
[    4.533659] SMP: stopping secondary CPUs                                     
[    4.537575] Kernel Offset: disabled                                          
[    4.541052] CPU features: 0x0002,20002004                                    
[    4.545044] Memory Limit: none                                               
[    4.548086] ---[ end Kernel panic - not syncing: VFS: Unable to mount root f-

yangye8, 2020-07-08:
Hi, sometimes you should be aware of Flashing results reported by Flash tools, it's not 100% right every time, so my suggestions are:
1. check your boot mode whether is SD boot
2. re-flash the image again and see the results reported by etcher tools
3. make sure SD card plug-in well

yangye8, 2020-07-08:
btw, not all SD cards are supported by Xilinx SD controller, usually, choose 8GB or 16GB SD card is a safe selection. Most 64GB or high capacity SD card are not supported by zcu102 SD controller. FYI

bettertm, 2020-07-08:
Thank you for your reply. I have found the cause of the error. Because the SD card has write protection, which leads to the above problems, just need to unlock the write protection.

closed 2020-07-08 by bettertm