Xilinx/Vitis-AI #114: when I setting up the Host,I meet this question?

tensorflowt, 2020-07-11:
`Your environment is misconfigured, you probably need to 'unset LD_LIBRARY_PATH'
but please check why this was set in the first place and that it's safe to unset.
The SDK will not operate correctly in most cases when LD_LIBRARY_PATH is set.
For more references see:
  http://tldp.org/HOWTO/Program-Library-HOWTO/shared-libraries.html#AEN80
  http://xahlee.info/UnixResource_dir/_/ldpath.html`

how can i solve it ?

thks

qianglin-xlnx, 2020-07-11:
Hi @tensorflowt 
Please follow the prompts to execute this command first :   unset LD_LIBRARY_PATH 
Then execute the following command again:
source <petalinux_sdk>/environment-setup-aarch64-xilinx-linux
Note that <petalinux_sdk> is the location where you install the sdk-2020.1.0.0.sh.

tensorflowt, 2020-07-11:
it's works,thank you.

closed 2020-07-13 by qianglin-xlnx