Xilinx/Vitis-AI #115: After I modified the library source code and compiled the Al libraries,how can I use the new libraries for inference in my task?

tensorflowt, 2020-07-11:
hi!

I followed the tutorial and modified the code in Vitis-AI-Library and compiled it, with the following results:

[image]

and I want to know,how can I use it for inference in my task.(i.e.,I need move this files in xxx path and compile it ?,or some way others)

thks

qianglin-xlnx, 2020-07-14:
Hi @tensorflowt 
For host cross compiling, you can put the new libraries under petalinux_sdk/sysroots/aarch64-xilinx-linux/usr/lib
For target board, you can copy the new libraries to the target via ssh and put them under /usr/lib in the target system.

tensorflowt, 2020-07-15:
Hi@qianglin-xlnx
I have tried according to your suggestion, but I have encountered the following mistakes. How can I solve them？
[image]
But when I go back to my old dynamic library, it works fine
[image]

I just replaced the following files in /usr/lib on board:

[image]

qianglin-xlnx, 2020-07-15:
Hi @tensorflowt 
For examples with video input, only webm and raw format are supported by default with the official system image. 
I notice that you use the rain_0.avi. Could you please check the video format again? If you install the ffmpeg packet in the system, please ignore this.

tensorflowt, 2020-07-15:
Hi @qianglin-xlnx
You may not understand me.Currently I need to modify the library code according to my own model, and then compile.
But when I tried to modify the code and compile it, the following problem occurred：

[image]

My implementation is as follows：

http://note.youdao.com/s/OXXhnqcw

How can I solved it ? Thanks.

qianglin-xlnx, 2020-07-21:
Hi @tensorflowt 
I have just looked at the link that you supplied. You install the Petalinux_sdk under ~ in the docker system. For docker system, we suggest to use the /workspace folder.  And for edge, there is no need to use docker system. You can install the sdk-2020.1.0.0.sh in your linux system. So could you can try it again? Thank you.

qianglin-xlnx, 2020-07-21:
@tensorflowt 
Could you try to install sdk-2020.1.0.0.sh in your linux system not in the docker system? And follow https://github.com/Xilinx/Vitis-AI/tree/master/Vitis-AI-Library#setting-up-the-host to setting up the host. 
Then before you modify the code, build the multitask sample first to make sure the host develop environment is ok. 
Finally, you can try to modify the code and replace the library that you modify.

tensorflowt, 2020-07-21:
Hi @qianglin-xlnx

ok！I will try it. thks

qianglin-xlnx, 2022-02-16:
Hi @tensorflowt 
Since we haven't received your reply for a long time, I'm going to close this issue. If you still have any questions, please feel free to reopen it. Thank you very much.

closed 2022-02-16 by qianglin-xlnx