Xilinx/Vitis-AI #117: Device tree in dpu-trd-bsp does not match the DPU-TRD

GkyHub, 2020-07-14:
I am using the vivado process to build a DPU project, I found that the offered petalinux bsp configures a 4-core DPU in system-user.dtsi while the TRD is a 2-core version. 

It seems that this does not affect the run of the pre-built samples. But when I install DNNDK on my own board, dexplorer reports "fail to open DPU device and exit...". I will check this problem later.

davidxudong, 2020-07-16:
Hi @GkyHub  

It is OK that the Device tree in dpu-trd-dsp does not math the default settings of DPU-TRD Vivado Flow.
As you mentioned it seems that this does not affect the run of the pre-built samples.

Dexplorer reports "fail to open DPU device and exit..."
I suggest you review your operation first.

Thanks

GkyHub, 2020-07-18:
Thanks, @davidxudong
I found that the dpu driver in the bsp is version 3.0 while the DNNDK in the mpsoc folder is for driver of version 4.0. I checked this by replacing the driver in the petalinux project with the latest version and dexplorer works well.

davidxudong, 2020-07-20:
@GkyHub Well Done

qianglin-xlnx, 2020-07-20:
Hi @GkyHub 
I'm going to close this. Please feel free to reopen if there are additional issues.

closed 2020-07-20 by qianglin-xlnx