Xilinx/Vitis-AI #118: DPU TRD Vivado flow - Petalinux not building

jalezeta, 2020-07-19:
Hello!
I am trying to implement the DPU TRD on a ZCU102 following the Vivado flow and I have recently enocuntered an issue while trying to build the Petalinux image using the pre builded HW configuration. Basically, the build log is complaining about not finding the zynqmp_latest file. I am following the instruction provided on the README.md under "DPU-TRD/prj/Vivado/README.md".
The build log is the following: 
```
jalezeta@ubuntu:~/Documents/Vitis-AI/DPU-TRD/prj/Vivado/dpu_petalinux_bsp/xilinx-zcu102-trd$ petalinux-build 
INFO: sourcing build tools
[INFO] building project
[INFO] sourcing build environment
[INFO] generating user layers
[INFO] generating workspace directory
INFO: bitbake petalinux-image-minimal
Parsing recipes: 100% |###############################################################| Time: 0:05:03
Parsing of 2963 .bb files complete (0 cached, 2963 parsed). 4232 targets, 167 skipped, 0 masked, 0 errors.
NOTE: Resolving any missing task queue dependencies
NOTE: Fetching uninative binary shim from file:///home/jalezeta/Documents/Vitis-AI/DPU-TRD/prj/Vivado/dpu_petalinux_bsp/xilinx-zcu102-trd/components/yocto/downloads/uninative/9498d8bba047499999a7310ac2576d0796461184965351a56f6d32c888a1f216/x86_64-nativesdk-libc.tar.xz;sha256sum=9498d8bba047499999a7310ac2576d0796461184965351a56f6d32c888a1f216
Initialising tasks: 100% |############################################################| Time: 0:00:11
Checking sstate mirror object availability: 100% |####################################| Time: 0:00:06
Sstate summary: Wanted 2570 Found 0 Missed 2570 Current 0 (0% match, 0% complete)
NOTE: Executing Tasks
NOTE: Setscene tasks completed
WARNING: dnndkdeploy-1.0-r0 do_fetch: Failed to fetch URL file:///proj/yocto/dnndk/zynqmp_latest, attempting MIRRORS if available
ERROR: dnndkdeploy-1.0-r0 do_fetch: Fetcher failure: Unable to find file file:///proj/yocto/dnndk/zynqmp_latest anywhere. The paths that were searched were:
    /home/jalezeta/Documents/Vitis-AI/DPU-TRD/prj/Vivado/dpu_petalinux_bsp/xilinx-zcu102-trd/components/yocto/layers/meta-vitis-ai/recipes-support/dnndk/dnndkdeploy-1.0/petalinux
    /home/jalezeta/Documents/Vitis-AI/DPU-TRD/prj/Vivado/dpu_petalinux_bsp/xilinx-zcu102-trd/components/yocto/layers/meta-vitis-ai/recipes-support/dnndk/dnndkdeploy/petalinux
    /home/jalezeta/Documents/Vitis-AI/DPU-TRD/prj/Vivado/dpu_petalinux_bsp/xilinx-zcu102-trd/components/yocto/layers/meta-vitis-ai/recipes-support/dnndk/files/petalinux
    /home/jalezeta/Documents/Vitis-AI/DPU-TRD/prj/Vivado/dpu_petalinux_bsp/xilinx-zcu102-trd/components/yocto/layers/meta-vitis-ai/recipes-support/dnndk/dnndkdeploy-1.0/zcu102-zynqmp
    /home/jalezeta/Documents/Vitis-AI/DPU-TRD/prj/Vivado/dpu_petalinux_bsp/xilinx-zcu102-trd/components/yocto/layers/meta-vitis-ai/recipes-support/dnndk/dnndkdeploy/zcu102-zynqmp
    /home/jalezeta/Documents/Vitis-AI/DPU-TRD/prj/Vivado/dpu_petalinux_bsp/xilinx-zcu102-trd/components/yocto/layers/meta-vitis-ai/recipes-support/dnndk/files/zcu102-zynqmp
    /home/jalezeta/Documents/Vitis-AI/DPU-TRD/prj/Vivado/dpu_petalinux_bsp/xilinx-zcu102-trd/components/yocto/layers/meta-vitis-ai/recipes-support/dnndk/dnndkdeploy-1.0/armv8a
    /home/jalezeta/Documents/Vitis-AI/DPU-TRD/prj/Vivado/dpu_petalinux_bsp/xilinx-zcu102-trd/components/yocto/layers/meta-vitis-ai/recipes-support/dnndk/dnndkdeploy/armv8a
    /home/jalezeta/Documents/Vitis-AI/DPU-TRD/prj/Vivado/dpu_petalinux_bsp/xilinx-zcu102-trd/components/yocto/layers/meta-vitis-ai/recipes-support/dnndk/files/armv8a
    /home/jalezeta/Documents/Vitis-AI/DPU-TRD/prj/Vivado/dpu_petalinux_bsp/xilinx-zcu102-trd/components/yocto/layers/meta-vitis-ai/recipes-support/dnndk/dnndkdeploy-1.0/aarch64
    /home/jalezeta/Documents/Vitis-AI/DPU-TRD/prj/Vivado/dpu_petalinux_bsp/xilinx-zcu102-trd/components/yocto/layers/meta-vitis-ai/recipes-support/dnndk/dnndkdeploy/aarch64
    /home/jalezeta/Documents/Vitis-AI/DPU-TRD/prj/Vivado/dpu_petalinux_bsp/xilinx-zcu102-trd/components/yocto/layers/meta-vitis-ai/recipes-support/dnndk/files/aarch64
    /home/jalezeta/Documents/Vitis-AI/DPU-TRD/prj/Vivado/dpu_petalinux_bsp/xilinx-zcu102-trd/components/yocto/layers/meta-vitis-ai/recipes-support/dnndk/dnndkdeploy-1.0/cortexa72-cortexa53
    /home/jalezeta/Documents/Vitis-AI/DPU-TRD/prj/Vivado/dpu_petalinux_bsp/xilinx-zcu102-trd/components/yocto/layers/meta-vitis-ai/recipes-support/dnndk/dnndkdeploy/cortexa72-cortexa53
    /home/jalezeta/Documents/Vitis-AI/DPU-TRD/prj/Vivado/dpu_petalinux_bsp/xilinx-zcu102-trd/components/yocto/layers/meta-vitis-ai/recipes-support/dnndk/files/cortexa72-cortexa53
    /home/jalezeta/Documents/Vitis-AI/DPU-TRD/prj/Vivado/dpu_petalinux_bsp/xilinx-zcu102-trd/components/yocto/layers/meta-vitis-ai/recipes-support/dnndk/dnndkdeploy-1.0/zynqmp
    /home/jalezeta/Documents/Vitis-AI/DPU-TRD/prj/Vivado/dpu_petalinux_bsp/xilinx-zcu102-trd/components/yocto/layers/meta-vitis-ai/recipes-support/dnndk/dnndkdeploy/zynqmp
    /home/jalezeta/Documents/Vitis-AI/DPU-TRD/prj/Vivado/dpu_petalinux_bsp/xilinx-zcu102-trd/components/yocto/layers/meta-vitis-ai/recipes-support/dnndk/files/zynqmp
    /home/jalezeta/Documents/Vitis-AI/DPU-TRD/prj/Vivado/dpu_petalinux_bsp/xilinx-zcu102-trd/components/yocto/layers/meta-vitis-ai/recipes-support/dnndk/dnndkdeploy-1.0/zynqmpeg
    /home/jalezeta/Documents/Vitis-AI/DPU-TRD/prj/Vivado/dpu_petalinux_bsp/xilinx-zcu102-trd/components/yocto/layers/meta-vitis-ai/recipes-support/dnndk/dnndkdeploy/zynqmpeg
    /home/jalezeta/Documents/Vitis-AI/DPU-TRD/prj/Vivado/dpu_petalinux_bsp/xilinx-zcu102-trd/components/yocto/layers/meta-vitis-ai/recipes-support/dnndk/files/zynqmpeg
    /home/jalezeta/Documents/Vitis-AI/DPU-TRD/prj/Vivado/dpu_petalinux_bsp/xilinx-zcu102-trd/components/yocto/layers/meta-vitis-ai/recipes-support/dnndk/dnndkdeploy-1.0/mali400
    /home/jalezeta/Documents/Vitis-AI/DPU-TRD/prj/Vivado/dpu_petalinux_bsp/xilinx-zcu102-trd/components/yocto/layers/meta-vitis-ai/recipes-support/dnndk/dnndkdeploy/mali400
    /home/jalezeta/Documents/Vitis-AI/DPU-TRD/prj/Vivado/dpu_petalinux_bsp/xilinx-zcu102-trd/components/yocto/layers/meta-vitis-ai/recipes-support/dnndk/files/mali400
    /home/jalezeta/Documents/Vitis-AI/DPU-TRD/prj/Vivado/dpu_petalinux_bsp/xilinx-zcu102-trd/components/yocto/layers/meta-vitis-ai/recipes-support/dnndk/dnndkdeploy-1.0/aarch64
    /home/jalezeta/Documents/Vitis-AI/DPU-TRD/prj/Vivado/dpu_petalinux_bsp/xilinx-zcu102-trd/components/yocto/layers/meta-vitis-ai/recipes-support/dnndk/dnndkdeploy/aarch64
    /home/jalezeta/Documents/Vitis-AI/DPU-TRD/prj/Vivado/dpu_petalinux_bsp/xilinx-zcu102-trd/components/yocto/layers/meta-vitis-ai/recipes-support/dnndk/files/aarch64
    /home/jalezeta/Documents/Vitis-AI/DPU-TRD/prj/Vivado/dpu_petalinux_bsp/xilinx-zcu102-trd/components/yocto/layers/meta-vitis-ai/recipes-support/dnndk/dnndkdeploy-1.0/
    /home/jalezeta/Documents/Vitis-AI/DPU-TRD/prj/Vivado/dpu_petalinux_bsp/xilinx-zcu102-trd/components/yocto/layers/meta-vitis-ai/recipes-support/dnndk/dnndkdeploy/
    /home/jalezeta/Documents/Vitis-AI/DPU-TRD/prj/Vivado/dpu_petalinux_bsp/xilinx-zcu102-trd/components/yocto/layers/meta-vitis-ai/recipes-support/dnndk/files/
    /home/jalezeta/Documents/Vitis-AI/DPU-TRD/prj/Vivado/dpu_petalinux_bsp/xilinx-zcu102-trd/build/downloads
ERROR: dnndkdeploy-1.0-r0 do_fetch: Fetcher failure for URL: 'file:///proj/yocto/dnndk/zynqmp_latest'. Unable to fetch URL from any source.
ERROR: Logfile of failure stored in: /home/jalezeta/Documents/Vitis-AI/DPU-TRD/prj/Vivado/dpu_petalinux_bsp/xilinx-zcu102-trd/build/tmp/work/zynqmp-xilinx-linux/dnndkdeploy/1.0-r0/temp/log.do_fetch.4934
ERROR: Task (/home/jalezeta/Documents/Vitis-AI/DPU-TRD/prj/Vivado/dpu_petalinux_bsp/xilinx-zcu102-trd/components/yocto/layers/meta-vitis-ai/recipes-support/dnndk/dnndkdeploy.bb:do_fetch) failed with exit code '1'
NOTE: Tasks Summary: Attempted 6995 tasks of which 0 didn't need to be rerun and 1 failed.

Summary: 1 task failed:
  /home/jalezeta/Documents/Vitis-AI/DPU-TRD/prj/Vivado/dpu_petalinux_bsp/xilinx-zcu102-trd/components/yocto/layers/meta-vitis-ai/recipes-support/dnndk/dnndkdeploy.bb:do_fetch
Summary: There was 1 WARNING message shown.
Summary: There were 2 ERROR messages shown, returning a non-zero exit code.

```

andrew0917, 2020-07-21:
We can fix this error in xilinx-zcu102-trd/project-spec/configs/config by adding Network sstate feeds URL, and there are two ways to realize it.

1. **Manually Modify** 
   modify xilinx-zcu102-trd/project-spec/configs/config file in bsp

   CONFIG_YOCTO_NETWORK_SSTATE_FEEDS_URL="http://petalinux.xilinx.com/sswreleases/rel-v${PETALINUX_VER%%.*}/aarch64/sstate-cache"

2. **PetaLinux Command Modify**
  step 1: input **petalinux-config** command in bsp
  step 2: add "http://petalinux.xilinx.com/sswreleases/rel-v${PETALINUX_VER%%.*}/aarch64/sstate-cache" 
[image]
[image]

andrew0917, 2020-07-21:
@jalezeta The DPU TRD Vivado flow have updated the bsp to fix this error, you can download the latest project and buiild it.

jalezeta, 2020-07-21:
Thank you @andrew0917! That solved the problem! I am happy to hear that the issue has been solved and the bsp has been updated accordingly. I will close this issue then.

closed 2020-07-21 by jalezeta