Xilinx/Vitis-AI #119: When I run the AI model on ZU104, how can I perform model performance analysis with B1152？

tensorflowt, 2020-07-21:
Do I need to specify the core type of the DPU during model compilation？If so, how do I change the compiled code？

[image]

qianglin-xlnx, 2020-07-21:
Hi @tensorflowt 
Before you start to evaluate the DPU with B1152, you have to design and integrate DPU IP in your system first.  You can refer to https://github.com/Xilinx/Vitis-AI/tree/master/DPU-TRD to do this. Note that the DPU IP in the official release system is B4096 not B1152.
During you design the DPU IP, you will get HWH file which is needed by DLet tool.
https://github.com/Xilinx/Vitis-AI/tree/master/DPU-TRD/prj/Vitis#522-get-hwh-file
Then use DLet tool to generate the new ZCU104.dcf file which is used by vai_c_caffe. 

So back to your question, you just need to update the ZCU104.dcf file.  
The following link will also help you to move on.
https://www.xilinx.com/html_docs/vitis_ai/1_2/gjp1592545687662.html

qianglin-xlnx, 2022-02-16:
Hi @tensorflowt 
Since we haven't received your reply for a long time, I'm going to close this issue. If you still have any questions, please feel free to reopen it. Thank you very much.

closed 2022-02-16 by qianglin-xlnx