Xilinx/Vitis-AI #120: ZCU102 board booting Error

yeliang2258, 2020-07-23:
I followed the method provided in the document and found the following problems after starting the ZCU102 board
[image]
I confirm that I have read and write permissions

qianglin-xlnx, 2020-07-23:
Hi @yeliang2258 ,
You can refer to https://github.com/Xilinx/Vitis-AI/issues/113 and check the following.
1. check your boot mode whether is SD boot
2. re-flash the image again and see the results reported by etcher tools
3. make sure SD card plug-in well
4. make sure the write protection of SD card is unlock 
5. not all SD cards are supported by Xilinx SD controller, usually, choose 8GB or 16GB SD card is a safe selection. Most 64GB or high capacity SD card are not supported by zcu102 SD controller.
6. make sure the download image file is complete. The MD5 are shown below.
c209c023327184ce787f90168d80aa31  xilinx-zcu102-dpu-v2020.1-v1.2.0.img.gz
fee0c35d467d6f2bb85b8913daabcbae  xilinx-zcu104-dpu-v2020.1-v1.2.0.img.gz
7. Make sure the SD card is stable. If possible, use another SD card and flash it again using ether tool.

yeliang2258, 2020-07-23:
I tried all the above methods, but it didn't work

yeliang2258, 2020-07-23:
> Hi @yeliang2258 ,
> You can refer to #113 and check the following.
>
> 1. check your boot mode whether is SD boot
> 2. re-flash the image again and see the results reported by etcher tools
> 3. make sure SD [...]

Thanks for replying. I tried all the above methods, but it didn't work

qianglin-xlnx, 2020-07-24:
@yeliang2258 
That's strange. We can't reproduce the problem you mentioned.
1. Could you post the results reported by ether tools? 
2. Does the ZCU102 board you used boot up successfully with the other system image? 
3. The detail information of SD card you used? Could you use another SD card and try it again?

hanxue, 2020-08-03:
I wonder if another SD card still produce this error message?

yeliang2258, 2020-08-03:
> I wonder if another SD card still produce this error message?

I have replaced the SD card and still the same error.

qianglin-xlnx, 2020-08-04:
> 
> 
> > I wonder if another SD card still produce this error message?
> 
> I have replaced the SD card and still the same error.

@yeliang2258  Could you please provide more detailed information, including every step and the result of your operation, so that we can reproduce the problem and find the root cause of the problem? Thank you very much.

yeliang2258, 2020-08-10:
> @yeliang2258 Could you please provide more detailed information, including every step and the result of your operation, so that we can reproduce the problem and find the root cause of the problem? [...]

I follow the steps provided by you step by step, the size of the USB disk is 16GB, the software used is balenaEtcher-1.5.45-x64, and the balenaEtcher shows Completed successfully.

andrew0917, 2020-08-10:
@yeliang2258  it is very detailed explanation as qianglin-xlnx described in  https://github.com/Xilinx/Vitis-AI/issues/120#issuecomment-662814905,beside, I only can think of two following aspects  for debug and analysis.
1 . Could you please access your burned SD card via Linux OS and verify if the ext4 partion is exist? if there are /bin,/etc/,/var,/tmp.....  under the ext4 partition?
2 . When you are using the  Etcher tool to burn SD card, there are three steps to finsh burninig, where in the 2 steps,you should click "change" button and confirm that you select the whole SD card partition, not only the part of all SD card partition.

[image]

closed 2020-09-03 by yeliang2258