Xilinx/Vitis-AI #121: Plate-Detection model training issue.

lllkiller-jlll, 2020-07-23:
Hi.

I'm trying to training plate-detection model of model zoo.

I read readme.md and I found prepare dataset section.

It told that ground truth file's format is below:
`
  imagename: 123342315_京DH09R8_20171201160736.jpg
  number of plates: 1
  coordinates of plate(clockwise order): 204 271 258 270 258 286 204 288
  image resize: resize 320 * 320.
`

I made groundtruth file like
`
SAN_FINEVu_X500_018.bmp
1
355 719 417 719 417 763 355 763
resize 320 * 320
SAN_FINEVu_X500_117.bmp
1
1693 644 1761 644 1761 676 1693 676
resize 320 * 320
.....
`

but I receive error message when I ran convert_direct_txt
`
E0723 00:14:34.590574  1136 convert_direct_txt.cpp:135] Total to be processed: 2.
E0723 00:14:34.590842  1136 io.cpp:145] Could not open or find file images/resize
E0723 00:14:34.672686  1136 convert_direct_txt.cpp:195] Processed 1 files.
`

I think resize line cause error.

So, I want to know how command reseize correctly.

Thanks.

lji72, 2020-07-23:
Hi, I think the error is caused by ground truth file.
The ground truth file must have three linefeed in single annotation of plate as following(without resize 320*320):
 SAN_FINEVu_X500_018.bmp
1
355 719 417 719 417 763 355 763

You should resize image to 320 * 320 with ground truth adaption before using convert_direct_txt command.

lllkiller-jlll, 2020-07-23:
@lji72 

Thanks for your reply.

I'm already trying to make lmdb with no resize ground truth.

It seems like making lmdb correctly.

But when I training with this lmdb, Xilinx-Caffe told me error:
`
F0722 23:52:52.184196  1091 data_transformer.cpp:1877] Check failed: img_height == height (1080 vs. 320)
`

I think that non-resized image file is reason.

So I think that input image have to resized at lmdb making time.

Could you tell me right way for training?

lji72, 2020-07-24:
Actually, now convert_direct_txt does not support resize image and ground truth, so you'd better resize image to 320 * 320 with ground truth adaption before using convert_direct_txt command. Then you can run training righly.

lllkiller-jlll, 2020-07-27:
Oh, I see..

That's unfortunate for me.

lllkiller-jlll, 2020-07-27:
If there are no other comments, I will close this issue after 3 days.

lji72, 2020-07-27:
yes, pls close this issue.

closed 2020-07-29 by lllkiller-jlll