Xilinx/Vitis-AI #122: Using DNNDK in with DPU TRD Vivado flow

jalezeta, 2020-07-23:
Hello there,

I am trying to build and use the DNNDK Runtime through the Linux low level drivers but I cannot run the cross compiled models. The following are the steps that I have successfully perfomed:

1. I have generated the DPU-TRD bitstream through the Vivado flow.
2. I have built the Petalinux image succesfully.
3. I have installed and sourced the sdk arm cross compiler.
4. I have modified the "Makefile.cfg" to disable Python, use the Linux low level drivers and set the compiler flags to use the sdk.sh cross compiler:
```
#--------------------------------------------------------------------------------------------
# Please choose one of the following PYTHON_CFG value for python support package
# This configuration is invalid when using sdk.sh
#--------------------------------------------------------------------------------------------
#PYTHON_CFG = yes
PYTHON_CFG = no

#--------------------------------------------------------------------------------------------
# Please choose one of the following AOL_PATH values and configure XRT or Linux
#--------------------------------------------------------------------------------------------
#AOL_PATH=./src/aol/xrt/
AOL_PATH=./src/aol/linux/

#--------------------------------------------------------------------------------------------
# Please choose one of the following CFLAGS values and configure UltraScale+ or Zynq 7000
# 1. for sdk.sh ----------------------------> CFLAGS += -O2 -DUSE_ARM_NEON
# 2. for gcc on arm64 target board ---------> CFLAGS += -O2 -DUSE_ARM_NEON
# 3. for gcc on arm32 target board ---------> CFLAGS += -O2 -mfloat-abi=hard -mfpu=neon -ffast-math -DUSE_ARM_NEON
# 4. for aarch64-linux-gnu -----------------> CFLAGS += -O2 -mcpu=cortex-a53 -DUSE_ARM_NEON
# 5. for arm-linux-gnueabihf ---------------> CFLAGS += -O2 -mcpu=cortex-a9 -mfloat-abi=hard -mfpu=neon -ffast-math -DUSE_ARM_NEON
#--------------------------------------------------------------------------------------------
CFLAGS += -O2 -DUSE_ARM_NEON
#CFLAGS += -O2 -DUSE_ARM_NEON
#CFLAGS += -O2 -mfloat-abi=hard -mfpu=neon -ffast-math -DUSE_ARM_NEON
#CFLAGS += -O2 -mcpu=cortex-a53 -DUSE_ARM_NEON
#CFLAGS += -O2 -mcpu=cortex-a9 -mfloat-abi=hard -mfpu=neon -ffast-math -DUSE_ARM_NEON
```
5. I have compiled and packaged the DNNDK.
6. I have installed DNNDK on the tarfet boar by installing it in the RootFS partition through:
`sudo ./install.sh /media/RootFS`
7. The intallation log outputs a "installation was successful message".
8. I compile the resnet50 example in the DNNDK sample folder targeting the zcu102.

The error comes at this point when I try to execute the cross compiled resnet50 model on the target zcu102 board. The output is the following:
`[DNNDK] fail to open DPU device and exit ...`

I don't know what might be causing this error. I have tested the resnet50 compiled model under the DPU-TRD project and it works. The problem is that the resnet50 app in the DPU-TRD uses VART and I am interested in using DNNDK since there is no source code available for VART and DNNDK code is available since VAI 1.2 release.

I guess that the Vivado flow supports legacy DNNDK through the Linux drivers instead of XRT. At least that is what I have unsderstood from the Vitis AI User guide:
"For legacy Vivado® basedDPU, it interacts with the underlying Linux DPU driver (instead of XRT) for DPU scheduling andresource management." https://www.xilinx.com/support/documentation/sw_manuals/vitis_ai/1_2/ug1414-vitis-ai.pdf

qianglin-xlnx, 2020-07-24:
@jalezeta 
If you want to use DNNDK in Vivado flow, please refer to https://github.com/Xilinx/Vitis-AI/tree/master/mpsoc/vitis_ai_dnndk_src to build the library and runtime driver.
So back to your issue, you have to build the dpu.ko and replace with it in your system.
https://github.com/Xilinx/Vitis-AI/tree/master/mpsoc/vitis_ai_dnndk_src#build-dpu-driver

jalezeta, 2020-07-27:
Thanks for the response @qianglin-xlnx. I have tried to follow the recommendations but I am not sure what the KERNELDIR path is. Is it referring to the kernel header files or the source files under the driver folder?

qianglin-xlnx, 2020-07-28:
@jalezeta 
The vitis-ai/mpsoc/vitis_ai_dnndk_src/driver has the same compilation and installation steps as other linux device drivers. You can compile it in two ways, one is to get the compiled linux kernel first, and then use this kernel to compile the driver module, so the KERNELDIR is the linux kernel which you used; 
the other is to add the driver source code to the BSP and compile it with the linux kernel. 
If you use the second method, the DPU driver code already exists in the BSP, and you need to update the relevant files. Please refer to Petalinux UG1144 Chapter 8 Customizing the Root File System for operation, Creating and Adding Custom Kernel Modules. Related issue https://github.com/Xilinx/Vitis-AI/issues/117

jalezeta, 2020-07-28:
I am trying to compile it as the regular linux device drivers with the already compiled linux kernel. My petalinux is the default configured with the BSP from https://github.com/Xilinx/Vitis-AI/tree/master/DPU-TRD/prj/Vivado. The problem comes when I execute the following command when I am in the Driver directory:
`make ARCH=arm64 CROSS_COMPILE=aarch64-linux-gnu- KERNELDIR=/path/to/your/kernel`
The output is the following one:
```
jalezeta@ubuntu:~/Documents/Vitis-AI/mpsoc/vitis_ai_dnndk_src/driver$ make ARCH=arm64 CROSS_COMPILE=aarch64-linux-gnu- KERNELDIR=/media/jalezeta/RootFS/lib/
KCFLAGS=" -DCONFIG_DPU_v1_3_0 -DSIG_BASE_ADDR=0X8FF00000 -Wno-error=date-time -Wno-date-time " make ARCH=arm64 CROSS_COMPILE=aarch64-linux-gnu- -C /media/jalezeta/RootFS/lib/ M=/home/jalezeta/Documents/Vitis-AI/mpsoc/vitis_ai_dnndk_src/driver modules
make[1]: Entering directory '/media/jalezeta/RootFS/lib'
make[1]: Nothing to be done for 'modules'.
make[1]: Leaving directory '/media/jalezeta/RootFS/lib'
```

jalezeta, 2020-08-06:
I am not sure about what KERNELDIR to use. I am using the prebuilt ZCU102 board image and I am pointing KERNELDIR to "/lib" but I am not sure if this is the correct path. I am stucked in this step.

andrew0917, 2020-08-07:
@jalezeta once you built the Petalinux image succesfully, your petalinux projct will download kernel source  and develop based on it,and the specific path is:  xilinx-zcu102-trd/build/tmp/work-shared/zcu102-zynqmp/kernel-source

jalezeta, 2020-08-07:
Thanks @andrew0917. I have tried to use that but first of all it requires to generate some config files that I could get through `make oldconfig && make prepare`. Then, the Kernel module compilation failed because some types.h files are not found.

```
jalezeta@ubuntu:~/Documents/Vitis-AI1.2.1/mpsoc/vitis_ai_dnndk_src/driver$ make ARCH=arm64 CROSS_COMPILE=aarch64-linux-gnu- KERNELDIR=~/Documents/Vitis-AI1.2.1/DPU-TRD/prj/Vivado/dpu_petalinux_bsp/xilinx-zcu102-trd/build/tmp/work-shared/zcu102-zynqmp/kernel-source/
KCFLAGS=" -DCONFIG_DPU_v1_3_0 -DSIG_BASE_ADDR=0X8FF00000 -Wno-error=date-time -Wno-date-time " make ARCH=arm64 CROSS_COMPILE=aarch64-linux-gnu- -C /home/jalezeta/Documents/Vitis-AI1.2.1/DPU-TRD/prj/Vivado/dpu_petalinux_bsp/xilinx-zcu102-trd/build/tmp/work-shared/zcu102-zynqmp/kernel-source/ M=/home/jalezeta/Documents/Vitis-AI1.2.1/mpsoc/vitis_ai_dnndk_src/driver modules
make[1]: Entering directory '/home/jalezeta/Documents/Vitis-AI1.2.1/DPU-TRD/prj/Vivado/dpu_petalinux_bsp/xilinx-zcu102-trd/build/tmp/work-shared/zcu102-zynqmp/kernel-source'
  CC [M]  /home/jalezeta/Documents/Vitis-AI1.2.1/mpsoc/vitis_ai_dnndk_src/driver/dpucore.o
In file included from ./include/linux/compiler.h:180:0,
                 from ./include/linux/export.h:42,
                 from ./include/linux/linkage.h:7,
                 from ./include/linux/kgdb.h:16,
                 from ./arch/arm64/include/asm/cacheflush.h:11,
                 from /home/jalezeta/Documents/Vitis-AI1.2.1/mpsoc/vitis_ai_dnndk_src/driver/dpucore.h:13,
                 from /home/jalezeta/Documents/Vitis-AI1.2.1/mpsoc/vitis_ai_dnndk_src/driver/dpucore.c:10:
./include/uapi/linux/types.h:5:10: fatal error: asm/types.h: No such file or directory
 #include <asm/types.h>
          ^~~~~~~~~~~~~
compilation terminated.
scripts/Makefile.build:265: recipe for target '/home/jalezeta/Documents/Vitis-AI1.2.1/mpsoc/vitis_ai_dnndk_src/driver/dpucore.o' failed
make[2]: *** [/home/jalezeta/Documents/Vitis-AI1.2.1/mpsoc/vitis_ai_dnndk_src/driver/dpucore.o] Error 1
Makefile:1652: recipe for target '/home/jalezeta/Documents/Vitis-AI1.2.1/mpsoc/vitis_ai_dnndk_src/driver' failed
make[1]: *** [/home/jalezeta/Documents/Vitis-AI1.2.1/mpsoc/vitis_ai_dnndk_src/driver] Error 2
make[1]: Leaving directory '/home/jalezeta/Documents/Vitis-AI1.2.1/DPU-TRD/prj/Vivado/dpu_petalinux_bsp/xilinx-zcu102-trd/build/tmp/work-shared/zcu102-zynqmp/kernel-source'
Makefile:43: recipe for target 'all' failed
make: *** [all] Error 2

```
 I suspect this is due to the cross compilation. Anyway, I don't understand in which concrete scenario the dpu.ko compilation command works (the one on the [README.md](https://github.com/Xilinx/Vitis-AI/tree/master/mpsoc/vitis_ai_dnndk_src) ). I am following the exact same steps and it does not work. The objective here is to compile the kernel module DPU.ko so I can use the low level linux libraries instead of the XRT. Another possibility that I see would be to create a module in the petalinux project through petalinux-create as explained in[ UG1144](https://www.xilinx.com/support/documentation/sw_manuals/xilinx2020_1/ug1144-petalinux-tools-reference-guide.pdf) in section Creating and Adding Custom Kernel Modules. Anyway, I did not found a way of creating a module from the source files present under the [driver](https://github.com/Xilinx/Vitis-AI/tree/master/mpsoc/vitis_ai_dnndk_src/driver) folder. 

I hope my problem is clearly explained and you can understand it.

andrew0917, 2020-08-10:
Hi jalezeta, as you describled, you want to build driver module via  kernel soure, the path that I provide in last commnt is kernel source  in petalinux poject, and the path of .config is xilinx-zcu102-trd/build/tmp/work/zcu102_zynqmp-xilinx-linux/linux-xlnx/5.4+gitAUTOINC+22b71b4162-r0/linux-zcu102_zynqmp-standard-build/.config, which is generated after you  configurate the kernel.
I have done a similiar work before, you can reference and try it. the specific steps as below:
Step 1: prepare your driver source and Makefile;
Step 2: copy the .config to the top directory of kernel source; 
Step 3: input the make command and the prapremeters that you want(ARCH,CROSS_COMPILE......)

This is my Makefile,you can reference it, and i think the command in your last comment can realize a similar function.
`obj-m:= target_name.o
CROSS_COMPILE:= aarch64-linux-gnu-
ARCH:= arm64
CC:= $(CROSS_COMPILE)gcc
LD:= $(CROSS_COMPILE)ld

CURRENT_PATH:=$(shell pwd)
LINUX_KERNEL_PATH:=/the/path/of/you/kernel/source

all:
        make -C ${LINUX_KERNEL_PATH} M=${CURRENT_PATH} modules 
clean:
        make -C ${LINUX_KERNEL_PATH} M=${CURRENT_PATH} clean`

As for [ README.md ](https://github.com/Xilinx/Vitis-AI/tree/master/mpsoc/vitis_ai_dnndk_src) and create a module in the petalinux project , I'm sorry that i'm not familiar with these ways.

jalezeta, 2020-08-21:
Thanks @andrew0917! 

I have closely followed the specific steps you have indicated, but the build is failing. Could you please tell me if you had any similar issue? I would highly appreciate that. This is the output log:

```
jalezeta@ubuntu:~/Desktop/driver$ make all 
make -C ~/Documents/Vitis-AI1.2.1/DPU-TRD/prj/Vivado/dpu_petalinux_bsp/xilinx-zcu102-trd/build/tmp/work-shared/zcu102-zynqmp/kernel-source M=/home/jalezeta/Desktop/driver modules
make[1]: Entering directory '/home/jalezeta/Documents/Vitis-AI1.2.1/DPU-TRD/prj/Vivado/dpu_petalinux_bsp/xilinx-zcu102-trd/build/tmp/work-shared/zcu102-zynqmp/kernel-source'
make[2]: *** No rule to make target '/home/jalezeta/Desktop/driver/dpu.o', needed by '__build'.  Stop.
Makefile:1652: recipe for target '/home/jalezeta/Desktop/driver' failed
make[1]: *** [/home/jalezeta/Desktop/driver] Error 2
make[1]: Leaving directory '/home/jalezeta/Documents/Vitis-AI1.2.1/DPU-TRD/prj/Vivado/dpu_petalinux_bsp/xilinx-zcu102-trd/build/tmp/work-shared/zcu102-zynqmp/kernel-source'
Makefile:22: recipe for target 'all' failed
make: *** [all] Error 2

```

My makefile the same as yours:
```

modname:=dpu
obj-m:=$(modname).o

PWD :=$(shell pwd)
MAKE :=make

CROSS_COMPILE:= aarch64-linux-gnu-
ARCH:= arm64
CC:= $(CROSS_COMPILE)gcc
LD:= $(CROSS_COMPILE)ld
KERNELDIR:=~/Documents/Vitis-AI1.2.1/DPU-TRD/prj/Vivado/dpu_petalinux_bsp/xilinx-zcu102-trd/build/tmp/work-shared/zcu102-zynqmp/kernel-source

all:
	make -C ${KERNELDIR} M=${PWD} modules

clean:
	make -C ${KERNELDIR} M=${PWD} clean
.PHONY: all clean

```
The driver folder where I am currently on is the same as in [driver](https://github.com/Xilinx/Vitis-AI/tree/master/mpsoc/vitis_ai_dnndk_src/driver) but I have modified the Makefile as you suggested.

andrew0917, 2020-08-24:
Hi jalezeta, please run **make ARCH=arm64 CROSS_COMPILE=aarch64-linux-gnu-**  or **export ARCH=arm64 CROSS_COMPILE=aarch64-linux-gnu- && make** command to generate *.ko file.

jalezeta, 2020-08-24:
Dear @andrew0917 I have tried what you have recommended. I thought the ARCH and CROSS_COMPILE should have been correct since they were specified in the Makefile. Anyway, I am still facing an error regarding the "types.h" file. Did you face this problem as well? Is it any dependency issue? The kernel version is "5.4.0-xilinx-v2020.1". 

```
jalezeta@ubuntu:~/Desktop/driver$ make ARCH=arm64 CROSS_COMPILE=aarch64-linux-gnu- 
make -C ~/Documents/Vitis-AI1.2.1/DPU-TRD/prj/Vivado/dpu_petalinux_bsp/xilinx-zcu102-trd/build/tmp/work-shared/zcu102-zynqmp/kernel-source M=/home/jalezeta/Desktop/driver modules
make[1]: Entering directory '/home/jalezeta/Documents/Vitis-AI1.2.1/DPU-TRD/prj/Vivado/dpu_petalinux_bsp/xilinx-zcu102-trd/build/tmp/work-shared/zcu102-zynqmp/kernel-source'
  CC [M]  /home/jalezeta/Desktop/driver/dpucore.o
In file included from ./include/linux/compiler.h:180:0,
                 from ./include/linux/export.h:42,
                 from ./include/linux/linkage.h:7,
                 from ./include/linux/kgdb.h:16,
                 from ./arch/arm64/include/asm/cacheflush.h:11,
                 from /home/jalezeta/Desktop/driver/dpucore.h:13,
                 from /home/jalezeta/Desktop/driver/dpucore.c:10:
./include/uapi/linux/types.h:5:10: fatal error: asm/types.h: No such file or directory
 #include <asm/types.h>
          ^~~~~
compilation terminated.
scripts/Makefile.build:265: recipe for target '/home/jalezeta/Desktop/driver/dpucore.o' failed
make[2]: * [/home/jalezeta/Desktop/driver/dpucore.o] Error 1
Makefile:1652: recipe for target '/home/jalezeta/Desktop/driver' failed
make[1]: * [/home/jalezeta/Desktop/driver] Error 2
make[1]: Leaving directory '/home/jalezeta/Documents/Vitis-AI1.2.1/DPU-TRD/prj/Vivado/dpu_petalinux_bsp/xilinx-zcu102-trd/build/tmp/work-shared/zcu102-zynqmp/kernel-source'
Makefile:22: recipe for target 'all' failed
make: * [all] Error 2
```

andrew0917, 2020-08-24:
Hi jalezeta,Although ARCH and CROSS_COMPILE are specified in the Makefile, but i found it does't work,so i recommended the way that i wrote in last comment. As for "types.h",i am sorry that i don't have this issue by flowing the steps that i describeded in https://github.com/Xilinx/Vitis-AI/issues/122#issuecomment-671148605.

jalezeta, 2020-08-26:
Hi @andrew0917, I have solved the "types.h" error by recompiling the kernel source and applying the default petalinux configuration through the default BSP under the DPU_TRD/Vivado project. Anyway, I am facing an error related to one flag of the aarch64 cross compiler. 

```
jalezeta@beluga:~/Documents/driver$ make ARCH=arm64 CROSS_COMPILE=aarch64-linux-gnu- KERNELDIR=~/Documents/kernel-source/
make -C  /home/jalezeta/Documents/kernel-source/ M=/home/jalezeta/Documents/driver modules
make[1]: Entering directory '/home/jalezeta/Documents/kernel-source'
  CC [M]  /home/jalezeta/Documents/driver/dpucore.o
aarch64-linux-gnu-gcc: error: missing argument to ‘-mstack-protector-guard-offset=’
make[2]: *** [scripts/Makefile.build:266: /home/jalezeta/Documents/driver/dpucore.o] Error 1
make[1]: *** [Makefile:1652: /home/jalezeta/Documents/driver] Error 2
make[1]: Leaving directory '/home/jalezeta/Documents/kernel-source'
make: *** [Makefile:15: all] Error 2

```

I have tried to modify the Makefile by including KCFLAGS and setting the "-mstack-protector-guard-offset=offset", but apparently I am setting the flag correctly. Do you have any idea on how to solve this? I am using Ubuntu 20.04. This is the Makefile:
```

obj-m:= dpucore.o
CROSS_COMPILE:= aarch64-linux-gnu-
ARCH:= arm64
CC:= $(CROSS_COMPILE)gcc
LD:= $(CROSS_COMPILE)ld

KCFLAGS=KCFLAGS="
KCFLAGS +=-mstack-protector-guard-offset=offset
KCFLAGS+="

CURRENT_PATH:=$(shell pwd)
KERNELDIR:=~/Documents/kernel-source/

all:
	make -C  ${KERNELDIR} M=${CURRENT_PATH} modules
clean:
	make -C ${KERNELDIR} M=${CURRENT_PATH} clean
```

andrew0917, 2020-08-28:
Hi jalezeta, I'm sorry  i don't have this issue. By referencing  your workflow,i generate the dpucore.ko successfully as below:
Step 1:  download the dpu source from [https://github.com/Xilinx/Vitis-AI/tree/master/mpsoc/vitis_ai_dnndk_src/driver](https://github.com/Xilinx/Vitis-AI/tree/master/mpsoc/vitis_ai_dnndk_src/driver)
Step 2:  follow the steps that i describeded in https://github.com/Xilinx/Vitis-AI/issues/122#issuecomment-671148605, and the whole operation log as below(by the way, my server environment is Ubuntu 18.04):

```
[My_server_hostname]$ git clone https://github.com/Xilinx/Vitis-AI.git
Cloning into 'Vitis-AI'...
remote: Enumerating objects: 18, done.
remote: Counting objects: 100% (18/18), done.
remote: Compressing objects: 100% (14/14), done.
remote: Total 30551 (delta 7), reused 12 (delta 4), pack-reused 30533
Receiving objects: 100% (30551/30551), 963.96 MiB | 14.38 MiB/s, done.
Resolving deltas: 100% (9839/9839), done.
Checking out files: 100% (22797/22797), done.
[My_server_hostname]$ cd Vitis-AI/mpsoc/vitis_ai_dnndk_src/driver
[My_server_hostname]$ ls
dpucore.c  dpucore.h  dpudef.h  Makefile
[My_server_hostname]$ cat Makefile-----------**Note: I have replaced the origin Makefile**
obj-m:= dpucore.o
CROSS_COMPILE:= aarch64-linux-gnu-
ARCH:= arm64
CC:= $(CROSS_COMPILE)gcc
LD:= $(CROSS_COMPILE)ld

CURRENT_PATH:=$(shell pwd)
#LINUX_KERNEL:=$(shell uname -r)
LINUX_KERNEL_PATH:=/local/workspace/junpengl/smartcamera/linux-xlnx_som

all:
        make -C ${LINUX_KERNEL_PATH} M=${CURRENT_PATH} modules
clean:
        make -C ${LINUX_KERNEL_PATH} M=${CURRENT_PATH} clean

[My_server_hostname]$ make ARCH=arm64 CROSS_COMPILE=aarch64-linux-gnu-
make -C /local/workspace/junpengl/smartcamera/linux-xlnx_som M=/local/workspace/junpengl/smartcamera/Vitis-AI/mpsoc/vitis_ai_dnndk_src/driver modules
make[1]: Entering directory '/local/workspace/junpengl/smartcamera/linux-xlnx_som'
  CC [M]  /local/workspace/junpengl/smartcamera/Vitis-AI/mpsoc/vitis_ai_dnndk_src/driver/dpucore.o
  Building modules, stage 2.
  MODPOST 1 modules
  CC [M]  /local/workspace/junpengl/smartcamera/Vitis-AI/mpsoc/vitis_ai_dnndk_src/driver/dpucore.mod.o
  LD [M]  /local/workspace/junpengl/smartcamera/Vitis-AI/mpsoc/vitis_ai_dnndk_src/driver/dpucore.ko
make[1]: Leaving directory '/local/workspace/junpengl/smartcamera/linux-xlnx_som'
[My_server_hostname]$ ls
dpucore.c  dpucore.ko   dpucore.mod.c  dpucore.o  Makefile       Module.symvers
dpucore.h  dpucore.mod  dpucore.mod.o  dpudef.h   modules.order
[My_server_hostname]$ lsb_release -a
LSB Version:    core-9.20170808ubuntu1-noarch:printing-9.20170808ubuntu1-noarch:security-9.20170808ubuntu1-noarch
Distributor ID: Ubuntu
Description:    Ubuntu 18.04.1 LTS
Release:        18.04
Codename:       bionic**

```

jalezeta, 2020-08-28:
Thanks @andrew0917 for your support. May I ask you what version of the aarch64-linux-gnu- are you using? I have tried what you said multiple times in Ubuntu 18.04 and 20.04 clean machines and still failing with the same error. I suspect this is an error caused by the Cross compiler. How did you install it? I used the apt package manager and installed the cross compiler through:

`$sudo apt-get install gcc-aarch64-linux-gnu` 

Is there anything that I am missing here? Otherwise I have no idea what is failing in the process.

andrew0917, 2020-08-28:
1 .The version of  aarch64-linux-gnu-xxx is **7.3.0**, the version details as below:
```
aarch64-linux-gnu-gcc -v
Using built-in specs.
COLLECT_GCC=aarch64-linux-gnu-gcc
COLLECT_LTO_WRAPPER=/usr/lib/gcc-cross/aarch64-linux-gnu/7/lto-wrapper
Target: aarch64-linux-gnu
Configured with: ../src/configure -v --with-pkgversion='Ubuntu/Linaro 7.3.0-16ubuntu3' --with-bugurl=file:///usr/share/doc/gcc-7/README.Bugs --enable-languages=c,ada,c++,go,d,fortran,objc,obj-c++ --prefix=/usr --with-gcc-major-version-only --with-as=/usr/bin/aarch64-linux-gnu-as --with-ld=/usr/bin/aarch64-linux-gnu-ld --program-suffix=-7 --enable-shared --enable-linker-build-id --libexecdir=/usr/lib --without-included-gettext --enable-threads=posix --libdir=/usr/lib --enable-nls --with-sysroot=/ --enable-clocale=gnu --enable-libstdcxx-debug --enable-libstdcxx-time=yes --with-default-libstdcxx-abi=new --enable-gnu-unique-object --disable-libquadmath --disable-libquadmath-support --enable-plugin --enable-default-pie --with-system-zlib --enable-multiarch --enable-fix-cortex-a53-843419 --disable-werror --enable-checking=release --build=x86_64-linux-gnu --host=x86_64-linux-gnu --target=aarch64-linux-gnu --program-prefix=aarch64-linux-gnu- --includedir=/usr/aarch64-linux-gnu/include
Thread model: posix
gcc version 7.3.0 (Ubuntu/Linaro 7.3.0-16ubuntu3)
[junpengl@xcdldmz190092master+driver]$ aarch64-linux-gnu-gcc --version
aarch64-linux-gnu-gcc (Ubuntu/Linaro 7.3.0-16ubuntu3) 7.3.0
Copyright (C) 2017 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```
2.  I install the Cross compiler via apt package manager,the same way with you.
`sudo apt-get install gcc-aarch64-linux-gnu`

3. I have show all the processes of how to build dpucore.ko for you, and i have no idea about the error that you face, if you still don't have a solution for it, I suggest you contact the author of [https://github.com/Xilinx/Vitis-AI/tree/master/mpsoc/vitis_ai_dnndk_src/driver](https://github.com/Xilinx/Vitis-AI/tree/master/mpsoc/vitis_ai_dnndk_src/driver) for how to build the dpucore.ko, all the steps tha I've described is based on my previous work of build driver from source, maybe there are some difference from my approach.

jalezeta, 2020-09-02:
> Hi jalezeta, as you describled, you want to build driver module via kernel soure, the path that I provide in last commnt is kernel source in petalinux poject, and the path of .config is [...]

Hello @andrew0917, I am still working on solving this error. It seems that we are following the same steps and building on the same environment. However, there is one step where I am not fully sure about my correctness. The step that I am talking about is step number 2, copying the ".config" file into the kernel-source root directory. The initial scenario just before copying the file is the following one:
_I am trying to compile the kernel module right after generating the Linux image, but it seems that the config file is missing. The output log is suggesting to hit "make oldconfig && make prepare" to solve it._

At this specific point, I copy the ".config" file to the root directory of the kernel-source and I directly hit the following command:

`make ARCH=arm64 CROSS_COMPILE=aarch64-linux-gnu-`

Still, I am getting the output log:

```
  ERROR: Kernel configuration is invalid.
include/generated/autoconf.h or include/config/auto.conf are missing.
 Run 'make oldconfig && make prepare' on kernel src to fix it.
```

Here I see two possible ways:

1. Hitting  "make prepare" in the kernel-source. I need to provide a lot of kernel configuration options on the terminal window. 
2. Or copying "auto.conf" and the ".config" file from "xilinx-zcu102-trd/build/tmp/work/zcu102_zynqmp-xilinx-linux/linux-xlnx/5.4+gitAUTOINC+22b71b4162-r0/linux-zcu102_zynqmp-standard-build/"

Sorry for being so insistent, but I am a little bit confused about this step. Otherwise, everything seems to be the same.

jalezeta, 2020-09-02:
> Dear @andrew0917,
> I have tried what you have recommended. I thought the ARCH and CROSS_COMPILE should have been correct since they were specified in the Makefile. Anyway, I am still facing an error [...]

I have solved the problem. Basically, these are the steps I have followed after copying the ".config" file into the kernel-source.

1. Change to the kernel-source directory:
`cd ~/Vitis-AI/DPU-TRD/prj/Vivado/xilinx-zcu102-trd/build/tmp/work-shared/zcu102-zynqmp/kernel-source`
2. Load the configuration (after copying the ".config" file to the kernel-source root directory):
`make ARCH=arm64 CROSS_COMPILE=aarch64-linux-gnu- menuconfig`
3. Exit the "Menuconfig"
4. Update the reference to "types.h" in "./include/uapi/linux/types.h" from 
`#include <asm/types.h>`
to
`#include <asm-generic/types.h>`
5. Go to the driver directory and build the driver.
`make ARCH=arm64 CROSS_COMPILE=aarch64-linux-gnu- KERNELDIR=~/Vitis-AI/DPU-TRD/prj/Vivado/xilinx-zcu102-trd/build/tmp/work-shared/zcu102-zynqmp/kernel-source`

This is what has solved my issue. Anyway, thanks @andrew0917 for your help!

closed 2020-09-02 by jalezeta

andrew0917, 2020-09-03:
Hi jalezeta, According to your solution, I guess the kernel configuration is the key for the issue that you faced. Anyway, you have solved the issue,well done!

jalezeta, 2020-09-03:
> Hi jalezeta, According to your solution, I think the kernel configuration is the key for the issue that you faced. Anyway, you have solved the issue,well done!

May I ask you about how you loaded the driver (dpu.ko)? Did you use insmod? I am getting an error in this step.

```
root@xilinx-zcu102-2020_1:/# insmod driver/dpu.o
insmod: ERROR: could not insert module driver/dpu.o: Invalid parameters

```