Xilinx/Vitis-AI #123: VAI_C Fatal

badcook, 2020-07-24:
when I try to compile cf_squeeze_imagenet in vitis-ai-1.0, I met this error:
[VAI_C][Fatal] Check failed for condition [layer_id < model.layer_size()] in [/home/dnnc/dnnc_impl/parser/caffe/caffeparser.cc:215] :Failed to find parameter for [prob] in caffemodel, make sure name in prototxt is consistent with that in caffemodel.
Is there any problem with the deploy.caffemodel and deploy.prototxt?

lllkiller-jlll, 2020-07-24:
Which Vtisi AI version are you using?

If you use Vitis AI v1.2 now, you should better use cf_squeeze_imagenet_227_227_0.76G_1.2 deployed by Xilinx Model Zoo v1.2.

badcook, 2020-07-24:
I use vitis-ai-1.0 and the model is from all_models_1.2

qianglin-xlnx, 2020-07-24:
@badcook 
Please try the following:
[image]
For more details, please refer to readme.md after you unzip the cf_squeeze_imagenet_227_227_0.76G_1.2.zip

closed 2020-07-27 by badcook