Xilinx/Vitis-AI #125: GPG error

badcook, 2020-07-27:
I try to build the GPU image locally but meet this problem:
Reading package lists...
W: GPG error: https://developer.download.nvidia.cn/compute/cuda/repos/ubuntu1804/x86_64  Release: The following signatures were invalid: BADSIG F60F4B3D7FA2AF80 cudatools <cudatools@nvidia.com>
E: The repository 'https://developer.download.nvidia.com/compute/cuda/repos/ubuntu1804/x86_64  Release' is not signed.

badcook, 2020-07-27:
I have re installed nvidia runtime but met another error:
E: Failed to fetch https://developer.download.nvidia.cn/compute/cuda/repos/ubuntu1804/x86_64/Packages.gz  File has unexpected size (214544 != 231929). Mirror sync in progress? [IP: 202.194.101.19 443]
   Hashes of expected file:
    - Filesize:231929 [weak]
    - SHA256:350d061b6bddecb163b1b76ad4d5122ee209d2435c1a572dcbc39be30d5ba33e
    - SHA1:3ddbfd93a78b5ac56ea654d8ffe6f2401aa4e9fe [weak]
    - MD5Sum:7b53ad7415ee9205aa0d3ea6aec7567d [weak]
   Release file created at: Fri, 24 Jul 2020 15:02:12 +0000
E: Some index files failed to download. They have been ignored, or old ones used instead.

hanxue, 2020-08-03:
The first error on `W: GPG error1` is because of invalid GPG key signature. It might be causing the 2nd error that you posted. 

First, delete off the `F60F4B3D7FA2AF80` key. 

```
sudo apt-key del F60F4B3D7FA2AF80
```

Then import the nvidia Ubuntu repo GPG key again

```
wget -qO - https://developer.download.nvidia.com/compute/cuda/repos/ubuntu1804/x86_64/7fa2af80.pub | sudo apt-key add -
```

This should solve the GPG error. 

One issue I have come across for commonly occuring file and download corruption is with the router / firewall. Check your Internet connection, or try to build the GPU Docker image on another network connection to identify if network is the cause of issue.

qianglin-xlnx, 2022-02-16:
HI @badcook 
Since we haven't received your reply for a long time, I'm going to close this issue. If you still have any questions, please feel free to reopen it. Thank you very much.