Xilinx/Vitis-AI #127: compile proxylessnas error

badcook, 2020-07-27:
I met this problem when compile proxylessnas into *.elf
[VAI_C-BACKEND][Check Failed: shift_cut >= 0][/home/dnnc/submodules/asicv2com/src/SlNode/SlNodeConv.cpp:83][DATA_OUTRANGE][Data value is out of range!]

qianglin-xlnx, 2020-07-31:
Hi @badcook
shift_cut = input_pos + weights_pos - output_pos 
And shift_cut  value must be >= 0. If shift_cut is <0, it can usually be one of the following reasons, could you please check the following.
1. If the original float model is not trained sufficiently, the accuracy is very poor, which leads to the unreasonable fixed-point position.
2. The original model training is very sufficient, but the use of false input images for fixed-point will also lead to unreasonable fixed-point position.
3. The quantization tool itself may have some bugs for new model, in this case, please contact your Xilinx FAE.

badcook, 2020-08-01:
Thank you so much! I will check my model @qianglin-xlnx

zxw602, 2020-12-18:
Hi @badcook ,

I met the same problem with you. Could you pls share me your updates? Have you already fixed it? Thanks!

Zxw

qianglin-xlnx, 2022-02-16:
Hi @badcook 
Since we haven't received your reply for a long time, we assume you have solved this issue and I'm going to close it. If you still have any questions, please feel free to reopen it. Thank you very much.

closed 2022-02-16 by qianglin-xlnx