Xilinx/Vitis-AI #129: Data value is out of range!

badcook, 2020-07-28:
why this happens when compiling model into *.elf in vai_c_tensorflow. Use B1024 on Ultra96V2
[VAI_C-BACKEND][Check Failed: kernel_param * input_channel_group <= weights_buf_depth][/home/dnnc/submodules/asicv2com/src/Operator/OperatorDptConv.cpp:60][DATA_OUTRANGE][Data value is out of range!]

qianglin-xlnx, 2020-07-29:
Hi @badcook 
The error shows that the layer of model is too big to fit into DPU. There are usually several solutions to this situation.
1. Use larger core DPU,  such as B1600, B2304 or B4096
2. Modify the model
3. Prune the model, reduce the input or output of channel

qianglin-xlnx, 2022-02-16:
Hi @badcook 
Since we haven't received your reply for a long time, I'm going to close this issue. If you still have any questions, please feel free to reopen it. Thank you very much.

closed 2022-02-16 by qianglin-xlnx