Xilinx/Vitis-AI #130: VAI_C_Caffe "No Front End Specified"

jalezeta, 2020-07-29:
I am trying to compile the pre-quantized model "cf_resnet50_imagenet_224_224_7.7G_1.2" from the AI Model Zoo and I am facing the following error:
```
(vitis-ai-caffe) jalezeta@ubuntu:/workspace/AI-Model-Zoo/cf_resnet50$ vai_c_caffe -p quantized/Edge/deploy.prototxt -c quantized/Edge/deploy.caffemodel -a arch/dpu-07-16-2020-18-15.json -o output/ -n resnet50
**************************************************
* VITIS_AI Compilation - Xilinx Inc.
**************************************************
**************************************************
*  NO FRONT END SPECIFIED
**************************************************
```
How can I specify the "Front end"?

qianglin-xlnx, 2020-07-29:
Hi @jalezeta 
Which platform do you use? And could you double check the dpu-07-16-2020-18-15.json file that you used.

When I compile the model for zcu102, the result is normal, as shown in the following. Maybe you can compile the model for zcu102 first to make sure the vai_c_caffe tool and model are ok. Then compile the model for your platform again. If you encounter the error again, you need to check the dpu-07-16-2020-18-15.json file and it's corresponding file <platform>.dcf file.  

vai_c_caffe --prototxt ./quantized/Edge/deploy.prototxt \
--caffemodel ./quantized/Edge/deploy.caffemodel \
--arch /opt/vitis_ai/compiler/arch/DPUCZDX8G/ZCU102/arch.json \
--output_dir model \
--net_name resnet50
[image]

jalezeta, 2020-07-29:
Thanks for clarifying @qianglin-xlnx! I was using the previous target naming instead of the "DPUCZDX8G" and that solved the "FRONT END" problem. Then I realized that I was not using the correct DCF path in the .json file. I was using the global path instead of the relative path in the Docker container (/workspace/YOUR/PATH). 

I can correctly generate the ELF file now.

closed 2020-07-29 by jalezeta