Xilinx/Vitis-AI #131: VI

jalezeta, 2020-07-29:


Mookel, 2020-07-30:
Hi, @jalezeta 
1. The DPU and Softmax are independent IPs and our compiler only targets for the DPU IPs;
2. The Softmax is not driven by instrucitons and our runtime will provide some necessary APIs to enable users to use the Softmax IP in black-box way.
In summary, from the compiler aspect, the Softmax layer is not supported by DPU and must be deployed in the CPU and users must use the APIs provided by the runtime to leverage the Softmax IP to accelerate their applications in the host deployment code. 
Hope it helps instead of confusion, thanks very much.

Mookel, 2020-07-30:
Hi, @jalezeta

The DPU and Softmax are independent IPs and our compiler only targets for the DPU IPs;
The Softmax is not driven by instrucitons and our runtime will provide some necessary APIs to enable users to use the Softmax IP in black-box way.
In summary, from the compiler aspect, the Softmax layer is not supported by DPU and must be deployed in the CPU and users must use the APIs provided by the runtime to leverage the Softmax IP to accelerate their applications in the host deployment code.
Hope it helps instead of confusion, thanks very much.

jalezeta, 2020-07-30:
Thanks for the reply @Mookel! Let's say I want to use my Softmax core in the DPU which as you said is not supported by the DPU, is it safe to say that I will only use the DPUkernel generated by VAI_C and completely substitute the CPUkernel from VAI_C with the API operations? My goal is to completely avoid using the CPUkernel from VAI_C and only use the DPUkernel.

Mookel, 2020-07-30:
> Thanks for the reply @Mookel! Let's say I want to use my Softmax core in the DPU which as you said is not supported by the DPU, is it safe to say that I will only use the DPUkernel generated by [...]

Yes, @jalezeta , the compiler will only generate instructions for the DPUKernel(s) and will do nothing for the CPUKernel(s). You can/must implement the CPUKernel in pure C++ code or using the APIs for Softmax IP. 

And it's worth noting that the dataflow of the whole model must be managed explicitly by host code, which means you have to get output tensors from the DPUKernel(also using APIs of runtime) after the DPUKernel is finished and feed them to the CPUKernel implemented by yourself. In general, the execution flow is:
Preprocessing --> DPUKernel Running --> CPUKernel Running --> PostProcessing.

closed 2020-08-27 by jalezeta