Xilinx/Vitis-AI #133: vai_c_tensorflow NO FRONT END SPECIFIED

PenroseZ, 2020-07-30:
I have the same problem in tesorflow. I think the dpuv2 in the. JSON file cannot be recognized after the Vitis ai1.2 version is updated. It can only recognize the name dpuczdx8g. Which version of DPU should I use if I want to run on the ultra 96 platform

qianglin-xlnx, 2020-07-31:
Hi @PenroseZ 
Starting from Vitis AI 1.2, a new DPU naming is used to better differentiate various DPUs served for different purposes. The old DPUv1/v2/v3 naming is deprecated. 
You should use DPUCZDX8G for edge devices, such as Ultra96, ZCU102 or ZCU104. 
You can refer to the similar issue: https://github.com/Xilinx/Vitis-AI/issues/130

qianglin-xlnx, 2022-02-16:
Hi @PenroseZ 
Since we haven't received your reply for a long time, we assume you have solved this issue and I'm going to close it. If you still have any questions, please feel free to reopen it. Thank you very much.

closed 2022-02-16 by qianglin-xlnx