Xilinx/Vitis-AI #134: DPU synthesis failed because of timing violation by Vitis

tkclimb, 2020-08-05:
I tried to run DPU build flow by simply following the instruction below. 
https://github.com/Xilinx/Vitis-AI/tree/master/DPU-TRD/prj/Vitis#52-build-and-run-flow

However, although I didn't touch the configuration, the build finally failed. it seems that hold time violation happened in the last step. 
Is there anyone who knows about this?
I put the terminal message below.

```
[12:27:45] Starting bitstream generation..
[12:31:39] Run vpl: Step impl: Failed
[12:31:39] Run vpl: FINISHED. Run Status: impl ERROR

===>The following messages were generated while  Compiling (bitstream) accelerator binary: dpu Log file: /home/yamada/tkhome/ghq/github.com/Xilinx/Vitis-AI/DPU-TRD/prj/Vitis/binary_container_1/link/vivado/vpl/prj/prj.runs/impl_1/runme.log  :
ERROR: [VPL-4] design did not meet timing - hold violation
ERROR: [VPL 60-773] In '/home/yamada/tkhome/ghq/github.com/Xilinx/Vitis-AI/DPU-TRD/prj/Vitis/binary_container_1/link/vivado/vpl/vivado.log', caught Tcl error:  problem implementing dynamic region, impl_1: route_design ERROR, please look at the run log file '/home/yamada/tkhome/ghq/github.com/Xilinx/Vitis-AI/DPU-TRD/prj/Vitis/binary_container_1/link/vivado/vpl/prj/prj.runs/impl_1/runme.log' for more information
WARNING: [VPL 60-732] Link warning: INFO: Platform does not require extraction of debug/profile metadata.
WARNING: [VPL 60-732] Link warning: No monitor points found for BD automation.
ERROR: [VPL 60-704] Integration error, problem implementing dynamic region, impl_1: route_design ERROR, please look at the run log file '/home/yamada/tkhome/ghq/github.com/Xilinx/Vitis-AI/DPU-TRD/prj/Vitis/binary_container_1/link/vivado/vpl/prj/prj.runs/impl_1/runme.log' for more information
ERROR: [VPL 60-1328] Vpl run 'vpl' failed
ERROR: [VPL 60-806] Failed to finish platform linker
INFO: [v++ 60-1442] [12:31:39] Run run_link: Step vpl: Failed
Time (s): cpu = 00:01:13 ; elapsed = 03:33:30 . Memory (MB): peak = 1340.699 ; gain = 0.000 ; free physical = 7880 ; free virtual = 50197
ERROR: [v++ 60-661] v++ link run 'run_link' failed
ERROR: [v++ 60-626] Kernel link failed to complete
ERROR: [v++ 60-703] Failed to finish linking
INFO: [v++ 60-1653] Closing dispatch client.
Makefile:70: recipe for target 'binary_container_1/dpu.xclbin' failed
make: *** [binary_container_1/dpu.xclbin] Error 1
```

qianglin-xlnx, 2020-08-05:
Hi @tkclimb 
If you have not modified Makefile, you can refer the chapter 5.4
https://github.com/Xilinx/Vitis-AI/tree/master/DPU-TRD/prj/Vitis#54-integrate-the-dpu-in-customer-platform
If you meet some timing issues. you can modify the [vivado] part of prj_config file (prop=run.impl_1.strategy=Performance_Explore) to change another implementation strategy and re-compile the project.
you can simply modify the /config_file/prj_confg. change the line44 prop=run.impl_1.strategy=Performance_Explore to prop=run.impl_1.strategy=Performance_NetDelay_low or prop=run.impl_1.strategy=Congestion_SpreadLogic_low and try again.

tkclimb, 2020-08-06:
@qianglin-xlnx 
Thanks, I understand. How should I resume the build without starting from the begging after I changed the configuration? Is there any make target for retry the "impl" (or "synth" if necessary)??

qianglin-xlnx, 2020-08-06:
@tkclimb 
There is no way. The project must re-build again. It calls the vivado to build the project actually.

tkclimb, 2020-08-11:
@qianglin-xlnx 
I tried to make the project with both strategies `Performance_NetDelay_low` and `Congestion_SpreadLogic_low` but they didn't work successfully. both case also failed by hold time violation.  
I put the error message in `binary_container_1/link/vivado/vpl/prj/prj.runs/impl_1/runme.log`

```
INFO: [Common 17-83] Releasing license: Implementation
342 Infos, 160 Warnings, 2 Critical Warnings and 0 Errors encountered.
route_design completed successfully
route_design: Time (s): cpu = 08:58:15 ; elapsed = 02:44:19 . Memory (MB): peak = 13644.312 ; gain = 1789.000 ; free physical = 7896 ; free virtual = 49438
source /home/yamada/tkhome/ghq/github.com/Xilinx/Vitis-AI/DPU-TRD/prj/Vitis/binary_container_1/link/vivado/vpl/scripts/_full_route_post.tcl
get_cells: Time (s): cpu = 00:00:05 ; elapsed = 00:00:05 . Memory (MB): peak = 13644.312 ; gain = 0.000 ; free physical = 7893 ; free virtual = 49435
get_cells: Time (s): cpu = 00:00:05 ; elapsed = 00:00:05 . Memory (MB): peak = 13644.312 ; gain = 0.000 ; free physical = 7880 ; free virtual = 49421
get_cells: Time (s): cpu = 00:00:05 ; elapsed = 00:00:05 . Memory (MB): peak = 13644.312 ; gain = 0.000 ; free physical = 7878 ; free virtual = 49420
INFO: [Timing 38-480] Writing timing data to binary archive.
Writing placer database...
Writing XDEF routing.
Writing XDEF routing logical nets.
Writing XDEF routing special nets.
Write XDEF Complete: Time (s): cpu = 00:01:26 ; elapsed = 00:00:29 . Memory (MB): peak = 13644.312 ; gain = 0.000 ; free physical = 6089 ; free virtual = 49194
INFO: [Common 17-1381] The checkpoint '/home/yamada/tkhome/ghq/github.com/Xilinx/Vitis-AI/DPU-TRD/prj/Vitis/binary_container_1/link/vivado/vpl/prj/prj.runs/impl_1/dr_routed_timing.dcp' has been generated.
write_checkpoint: Time (s): cpu = 00:02:30 ; elapsed = 00:01:27 . Memory (MB): peak = 13644.312 ; gain = 0.000 ; free physical = 7511 ; free virtual = 49387
report_timing_summary: Time (s): cpu = 00:00:33 ; elapsed = 00:00:16 . Memory (MB): peak = 13644.312 ; gain = 0.000 ; free physical = 7369 ; free virtual = 49244
report_timing_summary: Time (s): cpu = 00:00:47 ; elapsed = 00:00:18 . Memory (MB): peak = 13644.312 ; gain = 0.000 ; free physical = 7368 ; free virtual = 49244
ERROR: [runtcl-1] design did not meet timing - hold violation
INFO: [Timing 38-480] Writing timing data to binary archive.
Writing placer database...
Writing XDEF routing.
Writing XDEF routing logical nets.
Writing XDEF routing special nets.
Write XDEF Complete: Time (s): cpu = 00:01:26 ; elapsed = 00:00:29 . Memory (MB): peak = 13644.312 ; gain = 0.000 ; free physical = 5578 ; free virtual = 49017
INFO: [Common 17-1381] The checkpoint '/home/yamada/tkhome/ghq/github.com/Xilinx/Vitis-AI/DPU-TRD/prj/Vitis/binary_container_1/link/vivado/vpl/prj/prj.runs/impl_1/zcu102_base_wrapper_routed_error.dcp' has been generated.
write_checkpoint: Time (s): cpu = 00:02:31 ; elapsed = 00:01:28 . Memory (MB): peak = 13644.312 ; gain = 0.000 ; free physical = 6990 ; free virtual = 49199
ERROR: [Common 17-39] 'send_msg_id' failed due to earlier errors.

INFO: [Common 17-206] Exiting Vivado at Fri Aug  7 14:25:12 2020...
```

qianglin-xlnx, 2020-08-13:
> @qianglin-xlnx
> I tried to make the project with both strategies `Performance_NetDelay_low` and `Congestion_SpreadLogic_low` but they didn't work successfully. both case also failed by hold time [...]

Hi @tkclimb 
You can open the comment "#param=compiler.skipTimingCheckAndFrequencyScaling=1" in the prj_config file. it will not check the timing of project. 
But I wonder do you change anything of the platform or do you use the vitis 2020.1? There should be not the timing issues integrating 2 B4096 DPU.

tkclimb, 2020-08-13:
Hi, @qianglin-xlnx 
I'll try your suggestion, but I didn't change anything from the commit id `524f65224c52314155dafc011d488ed30e458fcb`.

aerovish30, 2020-10-03:
Hello, i am getting the same issue. I tried building the Vitis-AI using 2019.2 version. Vitis_Embedded_Platform_Source (checkout 2019.2) and then Vitis-AI (latest version).

qianglin-xlnx, 2020-10-09:
Hi @tkclimb @aerovish30 
Please modify the Makefile to use the prj_config, as shown in the following. It will integrate 2 DPU core and there will be no time violation issue. 

[image]

chhait07, 2021-05-21:
Hi @tkclimb ,
did you manage to solve the problem? I currently have the same issue. I want to compile the project with config_file prj_config_102_3dpu_LPD because I want to re-generate the standard Xilinx DPU-image for research purposes and I run into the same timing error as you wrote in your first post.

chhait07, 2021-05-24:
I tried different implementation strategies as @qianglin-xlnx proposed. For me with `Congestion_SpreadLogic_low` the DPU could successfully be built.

qianglin-xlnx, 2022-02-16:
Hi @tkclimb 
Since we haven't received your reply for a long time, I'm going to close this issue. If you still have any questions, please feel free to reopen it. Thank you very much.

closed 2022-02-16 by qianglin-xlnx