Xilinx/Vitis-AI #135: AI-Quantizer for pytorch failed to quantize the sample pretrained resnet18 model

tkclimb, 2020-08-06:
I tried to run quantization by following [the quick start](https://github.com/Xilinx/Vitis-AI/tree/master/Vitis-AI-Quantizer/vai_q_pytorch#quick-start-in-docker-environment).
the executed script is `Vitis-AI-Quantizer/vai_q_pytorch/example/resnet18_quant.py` on `xilinx/vitis-ai-gpu:latest` docker container and under `vitis-ai-pytorch` conda env.

While I succeeded the evaluation for the float model, the command is `python resnet18_quant.py --quant_mode 0`,
I encountered an error during the quantization with `--quant_mode 1 --subset_len 200` and `--quant_mode 1` as well.   
I put the message below. 

```shell
>> (vitis-ai-pytorch) yamada@F200713DT01:/workspace$ python resnet18_quant.py --quant_mode 1 --subset_len 200
[NNDCT_NOTE]: Loading NNDCT kernels...
-------- Start resnet18 test
[NNDCT_NOTE]: Quantization calibration process start up...
[NNDCT_NOTE]: =>Parsing ResNet...
Illegal instruction (core dumped)
```

I found the error happened at `torch.jit.get_trace_graph(model, args)` ([the source file](https://github.com/Xilinx/Vitis-AI/blob/master/Vitis-AI-Quantizer/vai_q_pytorch/pytorch_binding/pytorch_nndct/parse/utils.py#L160)), which is pytorch native function. I guess the given argument might be invalid but I haven't reached to the actual problem yet.

wk294701718, 2020-08-07:
i have the same  problem , my linux version is centos , whats your version

tkclimb, 2020-08-07:
@wk294701718 
my host os is ubuntu18.04 and the gpu docker container's environment is also ubuntu 18.04 (officially provided one).
I'm gonna try other framework example like tensorflow.

qianglin-xlnx, 2020-08-07:
Hi @tkclimb 
It seems that some instructions can't be recognized by your host.  So could you provide more details about your host?
The following is the information of my host which can run pytorch quantization normally. It's for your reference.

XXXXXXXXXXXXXX:~$ g++ --version
g++ (Ubuntu 7.5.0-3ubuntu1~18.04) 7.5.0
Copyright (C) 2017 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

XXXXXXXXXXXXXX:~$ cat /etc/issue
Ubuntu 18.04.1 LTS \n \l

XXXXXXXXXXXXXX:~$ uname -a
Linux XXXXXXXXXXXXXX 4.15.0-23-generic #25-Ubuntu SMP Wed May 23 18:02:16 UTC 2018 x86_64 x86_64 x86_64 GNU/Linux

The following the pytorch quantization running result in GPU docker.

(vitis-ai-pytorch) XXXXXXXXXXXXXX:/workspace/Vitis-AI-Quantizer/vai_q_pytorch/example$ python resnet18_quant.py --quant_mode 1 --subset_len 200

[NNDCT_NOTE]: Loading NNDCT kernels...
-------- Start resnet18 test

[NNDCT_NOTE]: Quantization calibration process start up...

[NNDCT_NOTE]: =>Parsing ResNet...

[NNDCT_NOTE]: =>Quantizable module is generated.(quantize_result/ResNet.py)
100%|#########################################| 7/7 [00:03<00:00,  2.14it/s]
loss: 0.042626
top-1 / top-5 accuracy: 69 / 88

[NNDCT_NOTE]: =>Exporting quant config.(quantize_result/quant_info.json)
-------- End of resnet18 test

dlphil, 2020-08-07:
I am also encountering this illegal instruction error, and have tried a number of possible solutions, including environment path changes and rebuilding the vitis-ai-gpu docker container using an nvidia cudnn10.2 image, in case the issue resulted from a CUDA mismatch. 

More debug detials can be read in this related Vitis-AI form thread: https://forums.xilinx.com/t5/AI-and-Vitis-AI/Pytorch-Illegal-instruction-core-dumped/td-p/1134124

@qianglin-xlnx, I'm encouraged to hear that you have it working. If possible, can you provide a print out of your environment settings in the docker container and information on the GPU and CUDA version you are using? Thank you.

dlphil, 2020-08-07:
@qianglin-xlnx, regarding your ask for host details, this is the pertinent info from my system:
(Ubuntu 7.5.0-3ubuntu1~18.04) 7.5.0
Ubuntu 18.04.4 LTS \n \l
Linux ip-XXXXXX #34~18.04.2-Ubuntu SMP Fri Jul 24 10:06:28 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux

For the CUDA version, can you provide the output for your system to the following commands?
cat /usr/local/cuda/version.txt
nvidia-smi

I see cuda version 10.0.130, and these are my driver details:
+-----------------------------------------------------------------------------+
| NVIDIA-SMI 440.33.01    Driver Version: 440.33.01    CUDA Version: 10.2     |
|-------------------------------+----------------------+----------------------+

Regarding the env paths in the container, I noticed that the LD_LIBRARY_PATH includes a tensorflow lib. I'm unsure if this is expected/required, but removing it had no effect. 

Thank you for your help.

dlphil, 2020-08-07:
For ease of reference, this is the gdb stack trace (as included in the forum thread: https://forums.xilinx.com/t5/AI-and-Vitis-AI/Pytorch-Illegal-instruction-core-dumped/m-p/1134864/highlight/true#M5090):

Starting program: /opt/vitis_ai/conda/envs/vitis-ai-pytorch/bin/python resnet18_quant.py --quant_mode 1 --subset_len 200
warning: Error disabling address space randomization: Operation not permitted
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[New Thread 0x7f82f32bf700 (LWP 552)]

[NNDCT_NOTE]: Loading NNDCT kernels...
-------- Start resnet18 test
[New Thread 0x7f82d49f8780 (LWP 553)]
Going into quantization...
This is the quant_mode 1

[NNDCT_NOTE]: Quantization calibration process start up...

[NNDCT_NOTE]: =>Parsing ResNet...

Thread 1 "python" received signal SIGILL, Illegal instruction.
**0x00007f8325a31cdb in mkldnn::impl::scales_t::set(int, int, float const*) () from /opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.6/site-packages/torch/lib/libcaffe2.so**

qianglin-xlnx, 2020-08-07:
> @qianglin-xlnx, regarding your ask for host details, this is the pertinent info from my system:
> (Ubuntu 7.5.0-3ubuntu1~18.04) 7.5.0 Ubuntu 18.04.4 LTS \n \l Linux ip-XXXXXX #34~18.04.2-Ubuntu SMP [...]

Hi @dlphil 
Following is the information of my system. 

:~$  cat /usr/local/cuda/version.txt
CUDA Version 10.2.89
:~$  nvidia-smi
Sat Aug  8 00:57:39 2020
+-----------------------------------------------------------------------------+
| NVIDIA-SMI 440.33.01    Driver Version: 440.33.01    CUDA Version: 10.2     |
|-------------------------------+----------------------+----------------------+

qianglin-xlnx, 2020-08-07:
> mkldnn::impl::scales_t::set

Maybe you can refer to the following to solve the issue about mkldnn.

https://github.com/pytorch/pytorch/issues/20583

dlphil, 2020-08-07:
Thank you, @qianglin-xlnx. For the CUDA version, is this within or outside the docker environment? 

By default, the DockerfileGPU includes
`FROM nvidia/cuda:10.0-cudnn7-runtime-ubuntu18.04`
so the CUDA version should be 10.0.130

I was able to get CUDA Version 10.2.89 within the container by switch the nvidia image to
`FROM nvidia/cuda:10.2-cudnn7-runtime-ubuntu18.04`
This did not resolve the illegal instruction error, though.

dlphil, 2020-08-07:
Yes, I saw this thread, as well, and confirmed that my system is using MKL-DNN v0.21.1 (https://github.com/oneapi-src/oneDNN/releases/tag/v0.21.1), which contains the fix. However, I noticed that the PyTorch version used by the docker environment does not.

`
>>> import torch
>>> print(*torch.__config__.show().split("\n"), sep="\n")
PyTorch built with:
  - GCC 7.3
  - Intel(R) Math Kernel Library Version 2020.0.1 Product Build 20200208 for Intel(R) 64 architecture applications
  - Intel(R) MKL-DNN v0.18.1 (Git Hash 7de7e5d02bf687f971e7668963649728356e0c20)
  - OpenMP 201511 (a.k.a. OpenMP 4.5)
  - NNPACK is enabled
  - CUDA Runtime 10.0
  - NVCC architecture flags: -gencode;arch=compute_30,code=sm_30;-gencode;arch=compute_35,code=sm_35;-gencode;arch=compute_50,code=sm_50;-gencode;arch=compute_52,code=sm_52;-gencode;arch=compute_60,code=sm_60;-gencode;arch=compute_61,code=sm_61;-gencode;arch=compute_70,code=sm_70;-gencode;arch=compute_75,code=sm_75;-gencode;arch=compute_52,code=compute_52
  - CuDNN 7.6.5
    - Built with CuDNN 7.3.1
  - Magma 2.5.0
  - Build settings: BLAS=MKL, BUILD_TYPE=Release, CXX_FLAGS=-march=nocona -mtune=haswell -ftree-vectorize -fPIC -fstack-protector-strong -fno-plt -O2 -ffunction-sections -pipe -I/opt/conda/conda-bld/pytorch_1565119753661/_h_env_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_place/include -fdebug-prefix-map=/opt/conda/conda-bld/pytorch_1565119753661/work=/usr/local/src/conda/pytorch-1.1.0 -fdebug-prefix-map=/opt/conda/conda-bld/pytorch_1565119753661/_h_env_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_place=/usr/local/src/conda-prefix  -D_FORTIFY_SOURCE=2 -O2 -I/opt/conda/conda-bld/pytorch_1565119753661/_h_env_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_place -Wno-deprecated -fvisibility-inlines-hidden -fopenmp -DUSE_FBGEMM -O2 -fPIC -Wno-narrowing -Wall -Wextra -Wno-missing-field-initializers -Wno-type-limits -Wno-array-bounds -Wno-unknown-pragmas -Wno-sign-compare -Wno-unused-parameter -Wno-unused-variable -Wno-unused-function -Wno-unused-result -Wno-strict-overflow -Wno-strict-aliasing -Wno-error=deprecated-declarations -Wno-stringop-overflow -Wno-error=pedantic -Wno-error=redundant-decls -Wno-error=old-style-cast -faligned-new -Wno-unused-but-set-variable -Wno-maybe-uninitialized -fno-math-errno -fno-trapping-math -Wno-stringop-overflow, DISABLE_NUMA=1, PERF_WITH_AVX=1, PERF_WITH_AVX2=1, PERF_WITH_AVX512=1, USE_CUDA=True, USE_EXCEPTION_PTR=1, USE_GFLAGS=OFF, USE_GLOG=OFF, USE_MKL=ON, USE_MKLDNN=ON, USE_MPI=OFF, USE_NCCL=True, USE_NNPACK=True, USE_OPENMP=ON,
`

dlphil, 2020-08-07:
@qianglin-xlnx, when possible, please run the following in Python from your docker container:
`
import torch
print(*torch.__config__.show().split("\n"), sep="\n")
`
If your version of MKL-DNN is v0.21, then there is a disparity between the docker environment currently pulled from release in GitHub and the version you are using.

qianglin-xlnx, 2020-08-10:
> 
> 
> @qianglin-xlnx, when possible, please run the following in Python from your docker container:
> 
> import torch
> print(*torch.**config**.show().split("\n"), sep="\n")
> (Referenced: https://discuss.pytorch.org/t/use-mkldnn-in-pytorch/54943)
> 
> If your version of MKL-DNN is v0.21, then there is a disparity between the docker environment currently pulled from release in GitHub and the version you are using.

Hi @dlphil 
Following is the information I got from the docker container. And the MKL-DNN is v0.18.1.

PyTorch built with:
  - GCC 7.3
  - Intel(R) Math Kernel Library Version 2020.0.1 Product Build 20200208 for Intel(R) 64 architecture applications
  - Intel(R) MKL-DNN v0.18.1 (Git Hash 7de7e5d02bf687f971e7668963649728356e0c20)
  - OpenMP 201511 (a.k.a. OpenMP 4.5)
  - NNPACK is enabled
  - CUDA Runtime 10.0
  - NVCC architecture flags: -gencode;arch=compute_30,code=sm_30;-gencode;arch=compute_35,code=sm_35;-gencode;arch=compute_50,code=sm_50;-gencode;arch=compute_52,code=sm_52;-gencode;arch=compute_60,code=sm_60;-gencode;arch=compute_61,code=sm_61;-gencode;arch=compute_70,code=sm_70;-gencode;arch=compute_75,code=sm_75;-gencode;arch=compute_52,code=compute_52
  - CuDNN 7.6.5
    - Built with CuDNN 7.3.1
  - Magma 2.5.0
  - Build settings: BLAS=MKL, BUILD_TYPE=Release, CXX_FLAGS=-march=nocona -mtune=haswell -ftree-vectorize -fPIC -fstack-protector-strong -fno-plt -O2 -ffunction-sections -pipe -I/opt/conda/conda-bld/pytorch_1565119753661/_h_env_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_place/include -fdebug-prefix-map=/opt/conda/conda-bld/pytorch_1565119753661/work=/usr/local/src/conda/pytorch-1.1.0 -fdebug-prefix-map=/opt/conda/conda-bld/pytorch_1565119753661/_h_env_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_place=/usr/local/src/conda-prefix  -D_FORTIFY_SOURCE=2 -O2 -I/opt/conda/conda-bld/pytorch_1565119753661/_h_env_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_place -Wno-deprecated -fvisibility-inlines-hidden -fopenmp -DUSE_FBGEMM -O2 -fPIC -Wno-narrowing -Wall -Wextra -Wno-missing-field-initializers -Wno-type-limits -Wno-array-bounds -Wno-unknown-pragmas -Wno-sign-compare -Wno-unused-parameter -Wno-unused-variable -Wno-unused-function -Wno-unused-result -Wno-strict-overflow -Wno-strict-aliasing -Wno-error=deprecated-declarations -Wno-stringop-overflow -Wno-error=pedantic -Wno-error=redundant-decls -Wno-error=old-style-cast -faligned-new -Wno-unused-but-set-variable -Wno-maybe-uninitialized -fno-math-errno -fno-trapping-math -Wno-stringop-overflow, DISABLE_NUMA=1, PERF_WITH_AVX=1, PERF_WITH_AVX2=1, PERF_WITH_AVX512=1, USE_CUDA=True, USE_EXCEPTION_PTR=1, USE_GFLAGS=OFF, USE_GLOG=OFF, USE_MKL=ON, USE_MKLDNN=ON, USE_MPI=OFF, USE_NCCL=True, USE_NNPACK=True, USE_OPENMP=ON,

dlphil, 2020-08-10:
Thank you, @qianglin-xlnx. 
It's curious that, even though we're using the same docker container with the outdated PyTorch and MKL-DNN, you're not encountering the mkldnn::impl::scales_t::set illegal instruction when running resnet18_quant.py with --quant_mode 1.
Since DockerfileGPU pulls PyTorch from a Xilinx repo, https://www.xilinx.com/bin/public/openDownload?filename=conda-channel_1.2.tar.gz , do you know if an updated conda-channel package exists, which could work with Vitis AI?
Thanks again.

tkclimb, 2020-08-11:
@qianglin-xlnx @dlphil 
Thanks for your information.
I compare my pytorch configuration with the above and it concludes completely same. I think it's not weird since we might use same docker container and conda environment.   
However, I noticed one point which my CPU is Ryzen3 (3950X). pytorch relays on I suspect there is a problem caused by the difference between intel CPU and AMD.  
which CPU is used on your machine?

qianglin-xlnx, 2020-08-11:
> 
> 
> @qianglin-xlnx @dlphil
> Thanks for your information.
> I compared my pytorch configuration with the above and it concludes completely same. I think it's not weird since we might use same docker container and conda environment.
> However, I noticed one point which my CPU is Ryzen3 (3950X). pytorch relays on MKL-DNN, therefore I suspect there is a problem caused by the difference between intel CPU and AMD.
> which CPU is used on your machine?

Hi @tkclimb 
Thank you for your valuable information. The CPU on my machine is Intel(R) Xeon(R) Gold 6154 CPU @ 3.00GHz.
MKL-DNN is an Intel acceleration library. I'm not sure if it can be used on AMD cpu. I also ask my colleague to take a closer look at the issue and see if MKL-DNN can be disabled.

niuxjxlnx, 2020-08-11:
hi, @tkclimb:

I suspect the conda channel to install pytorch maybe the root cause of the "illegal instruction" error. I tried to repeat the error on my servers but failed. So can you help verify the problem on your server with the following steps:

1. Create a clean conda environment

2. When install pytorch and torchvision, speicify conda channel with the following command line:
`conda install pytorch=1.1.0 torchvision=0.3.0 cudatoolkit=10.0 -c default`

3. Install pytorch_nndct(vai_q_torch) module from source code

4. run example case resnet18_quant.py.

The result of the upper steps shows whether pytorch in conda default channel works on your server or not. 

Then test pytorch in the official channel with the similar steps but only replace the channel specified in step 2 with the following command line:
`conda install pytorch=1.1.0 torchvision=0.3.0 cudatoolkit=10.0 -c pytorch`

tkclimb, 2020-08-11:
@niuxjxlnx 
Thanks.
I tried your suggestion and encountered a problem that ninja wasn't properly called. I'm not sure why it happened but I put the all commands I tried and the error messages on the `xilinx/vitis-ai-gpu:latest` docker container.

```shell
conda create -n clean-env # it's just a temporal name of my new conda environment
conda activate clean-env
conda install pytorch=1.1.0 torchvision=0.3.0 cudatoolkit=10.0 -c default # it's your suggestion

# from here, just followed the vai_q_pytorch official install flow from source
# https://github.com/Xilinx/Vitis-AI/tree/master/Vitis-AI-Quantizer/vai_q_pytorch#install-from-source-code)
export CUDA_HOME=/usr/local/cuda 
cd Vitis-AI-Quantizer/vai_q_pytorch
pip install -r requirements.txt # I needed this because setup.py install needs some additional python packages like scipy
cd pytorch_binding
python setup.py install
python setup.py develop
python -c "import pytorch_nndct" # the following error message appeared
```

The error message seems to say that cuda header might have been missing, or  ninja wasn't installed properly.  
I tried to run `ninja -v ' saying says the build file not found.
```
(clean-env) yamada@F200713DT01:/workspace/Vitis-AI-Quantizer/vai_q_pytorch/pytorch_binding$ python -c "import pytorch_nndct"
Traceback (most recent call last):
  File "/home/yamada/.conda/envs/clean-env/lib/python3.7/site-packages/torch/utils/cpp_extension.py", line 949, in _build_extension_module
    check=True)
  File "/home/yamada/.conda/envs/clean-env/lib/python3.7/subprocess.py", line 512, in run
    output=stdout, stderr=stderr)
subprocess.CalledProcessError: Command '['ninja', '-v']' returned non-zero exit status 1.

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/workspace/Vitis-AI-Quantizer/vai_q_pytorch/pytorch_binding/pytorch_nndct/__init__.py", line 55, in <module>
    import pytorch_nndct.apis
  File "/workspace/Vitis-AI-Quantizer/vai_q_pytorch/pytorch_binding/pytorch_nndct/apis/__init__.py", line 3, in <module>
    from .rnn_quant_api import LSTMQuantizer
  File "/workspace/Vitis-AI-Quantizer/vai_q_pytorch/pytorch_binding/pytorch_nndct/apis/rnn_quant_api.py", line 28, in <module>
    import pytorch_nndct.nn.modules.rnn_builder as rnn_buielder
  File "/workspace/Vitis-AI-Quantizer/vai_q_pytorch/pytorch_binding/pytorch_nndct/nn/__init__.py", line 1, in <module>
    from .modules import *
  File "/workspace/Vitis-AI-Quantizer/vai_q_pytorch/pytorch_binding/pytorch_nndct/nn/modules/__init__.py", line 2, in <module>
    from .linear import *
  File "/workspace/Vitis-AI-Quantizer/vai_q_pytorch/pytorch_binding/pytorch_nndct/nn/modules/linear.py", line 6, in <module>
    from .fix_ops import NndctScale
  File "/workspace/Vitis-AI-Quantizer/vai_q_pytorch/pytorch_binding/pytorch_nndct/nn/modules/fix_ops.py", line 4, in <module>
    from ..load_kernels import nndct_kernels
  File "/workspace/Vitis-AI-Quantizer/vai_q_pytorch/pytorch_binding/pytorch_nndct/nn/load_kernels.py", line 53, in <module>
    extra_include_paths=extra_include_paths)
  File "/home/yamada/.conda/envs/clean-env/lib/python3.7/site-packages/torch/utils/cpp_extension.py", line 644, in load
    is_python_module)
  File "/home/yamada/.conda/envs/clean-env/lib/python3.7/site-packages/torch/utils/cpp_extension.py", line 813, in _jit_compile
    with_cuda=with_cuda)
  File "/home/yamada/.conda/envs/clean-env/lib/python3.7/site-packages/torch/utils/cpp_extension.py", line 866, in _write_ninja_file_and_build
    _build_extension_module(name, build_directory, verbose)
  File "/home/yamada/.conda/envs/clean-env/lib/python3.7/site-packages/torch/utils/cpp_extension.py", line 962, in _build_extension_module
    raise RuntimeError(message)
RuntimeError: Error building extension 'nndct_kernels': b"[1/4] /usr/local/cuda/bin/nvcc -DTORCH_EXTENSION_NAME=nndct_kernels -DTORCH_API_INCLUDE_EXTENSION_H -I/workspace/Vitis-AI-Quantizer/vai_q_pytorch/include -I/workspace/Vitis-AI-Quantizer/vai_q_pytorch/pytorch_binding/pytorch_nndct/nn/include -isystem /home/yamada/.conda/envs/clean-env/lib/python3.7/site-packages/torch/include -isystem /home/yamada/.conda/envs/clean-env/lib/python3.7/site-packages/torch/include/torch/csrc/api/include -isystem /home/yamada/.conda/envs/clean-env/lib/python3.7/site-packages/torch/include/TH -isystem /home/yamada/.conda/envs/clean-env/lib/python3.7/site-packages/torch/include/THC -isystem /usr/local/cuda/include -isystem /home/yamada/.conda/envs/clean-env/include/python3.7m -D_GLIBCXX_USE_CXX11_ABI=1 -D__CUDA_NO_HALF_OPERATORS__ -D__CUDA_NO_HALF_CONVERSIONS__ -D__CUDA_NO_HALF2_OPERATORS__ --compiler-options '-fPIC' -std=c++11 -c /workspace/Vitis-AI-Quantizer/vai_q_pytorch/csrc/nndct_cuda_math.cu -o nndct_cuda_math.cuda.o\nFAILED: nndct_cuda_math.cuda.o \n/usr/local/cuda/bin/nvcc -DTORCH_EXTENSION_NAME=nndct_kernels -DTORCH_API_INCLUDE_EXTENSION_H -I/workspace/Vitis-AI-Quantizer/vai_q_pytorch/include -I/workspace/Vitis-AI-Quantizer/vai_q_pytorch/pytorch_binding/pytorch_nndct/nn/include -isystem /home/yamada/.conda/envs/clean-env/lib/python3.7/site-packages/torch/include -isystem /home/yamada/.conda/envs/clean-env/lib/python3.7/site-packages/torch/include/torch/csrc/api/include -isystem /home/yamada/.conda/envs/clean-env/lib/python3.7/site-packages/torch/include/TH -isystem /home/yamada/.conda/envs/clean-env/lib/python3.7/site-packages/torch/include/THC -isystem /usr/local/cuda/ [... cut ...] tis-AI-Quantizer/vai_q_pytorch/csrc/nndct_cu_utils.cc:19:0:\n/workspace/Vitis-AI-Quantizer/vai_q_pytorch/csrc/../include/nndct_cu_utils.h:21:10: fatal error: cuda.h: No such file or directory\n #include <cuda.h>\r\n          ^~~~~~~~\ncompilation terminated.\nninja: build stopped: subcommand failed.\n"

(clean-env) yamada@F200713DT01:/workspace/Vitis-AI-Quantizer/vai_q_pytorch/pytorch_binding$ ninja -v  # just in case I run ninja
ninja: error: loading 'build.ninja': No such file or directory
```

tkclimb, 2020-08-11:
> @qianglin-xlnx @dlphil
> Thanks for your information.
> I compared my pytorch configuration with the above and it concludes completely same. I think it's not weird since we might use same docker [...]

I tried `python resnet18_quant.py --quant_mode 1 --subset_len 200` on DGX-Station but it failed with the same illegal instruction error, so maybe it't not the cpu related problem.  
for your information, DGX station's cpu is Intel(R) Xeon(R) CPU E5-2698

niuxjxlnx, 2020-08-11:
@tkclimb 
When installing vai_q_pytorch from source code, it should only do either "python setup.py install" or "python setup.py develop", but can not do the two command lines both.

niuxjxlnx, 2020-08-11:
@tkclimb

niuxjxlnx, 2020-08-12:
It seems that we got the reason of the problem, please check the link https://forums.xilinx.com/t5/AI-and-Vitis-AI/Pytorch-Illegal-instruction-core-dumped/td-p/1134124.

domdal, 2020-11-25:
@tkclimb have you found a solution to the problem?
I have tried it from a clean linux Installation with these instruction - https://github.com/Xilinx/Vitis-AI/tree/master/Vitis-AI-Quantizer/vai_q_pytorch#install-from-source-code
and I am running into the same error as you. 
Im running on a Intel(R) Xeon(R) CPU E5-2690

TenzinCHW, 2020-11-26:
Hi, I am able to run the resnet example with `--quant_mode 2` on Google Colab. You'll need to go to dashboard.ngrok.com/auth (set up an account first and get the authentication token from that link). Since you also need the ImageNet dataset, I have a "small" sample [here](https://drive.google.com/file/d/1Q1m1tzSgiZpXn3H_tnahcWydYmCVhXsQ/view?usp=sharing), which you should add to your Google Drive home and then mount the drive (click the folder icon on the left in Colab and then click the Google Drive icon next to the Refresh icon). I do not know if I'm allowed to share this small sample of 5 images per class so I may have to take it down (anyone who knows?).

First, I run this in a cell:
```
!pip install colab_ssh
from colab_ssh import launch_ssh
auth_token = 'your-auth-token' # replace this string with token
pw = 'verysecretpassword' # set up a password if you want
launch_ssh(auth_token, pw)
```

It will print the `HostName` and `Port` number. Go to a terminal and ssh into the host with:
`ssh root@<HostName> -p <Port>`
and enter the password you set up above.

Copy and paste the following to set up miniconda:
```
wget https://repo.anaconda.com/miniconda/Miniconda3-latest-Linux-x86_64.sh -O miniconda3.sh
bash miniconda3.sh -b -p ~/conda
wget -O conda-channel.tar.gz https://www.xilinx.com/bin/public/openDownload?filename=conda-channel_1.2.tar.gz
tar -xzvf conda-channel.tar.gz
~/conda/bin/conda init bash
exit
```

`ssh` back in and then run the following (following instructions from the forum post shared by @niuxjxlnx):
```
cp /content/drive/MyDrive/smolImageNet.zip .
unzip smolImageNet.zip
wget https://download.pytorch.org/models/resnet18-5c106cde.pth -O resnet18.pth
git clone https://github.com/Xilinx/Vitis-AI.git
conda create -n vitis-ai-pytorch python=3.6 pytorch=1.1.0 torchvision=0.3.0 cudatoolkit=10.0 -y -c pytorch
conda deactivate
conda activate vitis-ai-pytorch
conda install xir -y -c ~/conda-channel
pip install -r Vitis-AI/Vitis-AI-Quantizer/vai_q_pytorch/requirements.txt
cd Vitis-AI/Vitis-AI-Quantizer/vai_q_pytorch/pytorch_binding
python setup.py install
cd ~
python ~/Vitis-AI/Vitis-AI-Quantizer/vai_q_pytorch/example/resnet18_quant.py --data_dir ./smolImageNet --model_dir . --quant_mode 1
python ~/Vitis-AI/Vitis-AI-Quantizer/vai_q_pytorch/example/resnet18_quant.py --data_dir ./smolImageNet --model_dir . --quant_mode 2
```

And you should be able to find the compiled model in the `~/quantize_result` directory. Sorry there are so many steps, most of the machines I've tried this on did not have the CUDA version required (10.0) and I was afraid to update them because they're shared servers, so I resorted to Colab. Hope this helps someone.

TenzinCHW, 2020-12-01:
I realised the above steps are a bit too many steps so I created a [Colab file](https://colab.research.google.com/drive/1ZqQVlxWOrH5up1DgioEhMUE2fAPSbpdx?usp=sharing) while does all the setup. However you still need to allocate the GPU runtime and upload data (change lines 5-6 of cell 3 if you have your own data). The quantized model file should be in `quantize_result/ResNet_int.xmodel`.

The relevant steps are a clean install of conda, installation of pytorch/cudatoolkit from pytorch channel, correct CUDA version (the one being used on Colab is 10.0).

qianglin-xlnx, 2022-02-16:
HI @tkclimb 
Since we haven't received your reply for a long time, we assume you have solved this issue and I'm going to close it. If you still have any questions, please feel free to reopen it. Thank you very much.

closed 2022-02-16 by qianglin-xlnx