Xilinx/Vitis-AI #136: VITIS-AI for windows?

kevinkit, 2020-08-08:
Hello,

I am wondering if VITIS-AI is limited to linux distributions only. Since it is docker based it should be possible to execute in on Windows - however the docker_run.sh files are Linux specific. 

Is there any plan to get vitis-ai on windows or am I missing something?

hanxue, 2020-09-07:
Hi @kevinkit , 

Vitis AI is distributed as Docker images - [any operating system that supports Docker ](https://github.com/Xilinx/Vitis-AI/blob/master/doc/install_docker/README.md)(specifically Docker 19.03.1 and above) can run Vitis AI. I have been able to run Vitis AI on a Windows 10 laptop with Docker for Windows installed. 

1. Install [Docker for Windows](https://docs.docker.com/docker-for-windows/install/)
2. Start Docker Desktop
3. Open up cmd, powershell or any Windows terminal, and you will be able to run all the `docker` commands

kevinkit, 2020-09-07:
Hey, 

Thank you for your reply  - I am little bit confused : In the docker_run.sh file https://github.com/Xilinx/Vitis-AI/blob/master/docker_run.sh 

the docker run commands are very unix specific using folders that do not exist in windows like /opt/xilinx/overlaybins etc. will this cause any trouble?

hanxue, 2020-09-07:
When you are running Docker for Windows, there is a thin layer of Linux, specifically Linux Container Support running on top of Windows. 

[image]

In short, when using Docker for Windows and using the `docker` commands on Linux-based Docker images, you are running in a virtualized Linux environment. 

Paths such as /opt/xilinx/overlaybins and /workspace are available within the virtual Linux environment in Docker for Windows; they will not be mapped to the host Windows filesystem. Therefore, you do not have to worry about Linux file paths in Vitis AI not working on a Windows 10 machine running Docker.

kevinkit, 2020-09-07:
Thank you really much for that explanation! This helps a lot, sadly I am forced to work with a very old windows version that is not supported by the newest docker file - looking forward for an upgrade !

closed 2020-09-07 by hanxue

kevinkit, 2020-09-21:
It worked under windows with cpu docker, however it seems to be a little bit harder with gpu - how do I know my device id  that i need to parse? What can i do if it does not find the nvidia runtime?

daivikbhatia, 2020-11-09:
Hi, How did you make it work for CPU? I installed latest version of docker, pulled the Vitis-AI repository from git in git bash and followed the rest of instructions,
I got this error while running docker:
find: ‘/dev/dri’: No such file or directory

and this error while building CPU docker:
failed to solve with frontend dockerfile.v0: failed to resolve dockerfile: failed to build LLB: error from sender: DockerfileCPU is not a directory

Please help!

hanxue, 2020-11-17:
@kevinkit just noticed your follow up question. Unfortunately nvidia-docker is not supported in Windows, and as far as I understand there is no plans to make nvidia-docker available on Windows https://github.com/NVIDIA/nvidia-docker/wiki/Frequently-Asked-Questions#is-microsoft-windows-supported

kevinkit, 2020-11-17:
@hanxue thank you, i got some success with the fact that windows is now supporting the linux subsystem, but it is so long ago that i do not quit recall it

krishnajy, 2020-12-04:
Openvino  AI  system  works on windows 10 .  Uses Docker , Python , visual studio , builder c/c++ ,  tensorflow .. .....