Xilinx/Vitis-AI #138: Support for ZCU-106

kevinkit, 2020-08-10:
Are there any plans to support the ZCU-106 Board in the near future?

wilderfield, 2020-08-10:
While not prepackaged, with enough elbow grease, anyone can run Vitis AI apps on any Zynq UltraScale+ board. One needs to build the base platform with Vivado, build the firmware and root file system with PetaLinux, and insert the DPU accelerator + build bitstream with Vitis.

kevinkit, 2020-08-18:
That is very sad to hear cause this will lead to a very bad user experience. VITIS-Ai seems to be a very interesting framework which has a lot potential, but it seems wasted limiting the 'easy experience' to two boards. While @jimheaton provided a nice port of the ZCU106 using the ZCU104, it does not support v 1.2 - it seems like using any other board then the ZCU102 or ZCU104 will always leave you with more work or behind current updates - this will most likely scare off new users. Imagine coming from Python like most AI Users nowadays or using something like the Jetsonboard and now one has to face this huge challenges to get something like the model zoo up and running - this will lead to a very bad user experience, sadly.

andyluo7, 2020-08-18:
@kevinkit, we'd like to support more different boards, but we have to select a few boards to support as default platforms in Vitis AI. ZCU104 has the same devices as ZCU106 with a lower price. Vitis AI also supports low-cost platform such as Avnet Ultra96v2. https://www.hackster.io/AlbertaBeef/vitis-ai-1-1-flow-for-avnet-vitis-platforms-part-1-007b0e. We will have detailed tutorials to show how to port Vitis AI to custom platforms soon.

kevinkit, 2020-08-19:
This is great news! Thank you Is there a schedule when these tutorials will come? 

I also fund this tutorial - can this be used for porting Vitis AI 1.2 to the ZCU106 or is this "too far" away from the actual problem?

https://github.com/hill19850213/vitis_ai_custom_platform_v1.2.1

kevinkit, 2020-09-04:
@andyluo7 any update on this?

jimheaton, 2020-09-29:
If you are willing to use Vitis-AI 1.1, you can use this: https://github.com/jimheaton/Vitis-AI-DPU_TRD-for-ZCU106
I have not updated this to 1.2 yet

andyluo7, 2021-02-26:
@kevinkit , please check https://github.com/Xilinx/Vitis-In-Depth-Tutorial/blob/master/Vitis_Platform_Creation/Introduction/02-Edge-AI-ZCU104/README.md

closed 2021-02-26 by andyluo7