Xilinx/Vitis-AI #139: Dependency on unversioned so file

paroque28, 2020-08-13:
While running the lanedetect demo I ran into this error:
```
bash-5.0# test_jpeg_lanedetect                                                                                                                                                            
terminate called after throwing an instance of 'std::logic_error'
  what():  basic_string::_M_construct null not valid
Aborted (core dumped)
bash-5.0# test_jpeg_lanedetect vpgnet_pruned_0_99 sample_lanedetect.jpg
WARNING: Logging before InitGoogleLogging() is written to STDERR
F0813 00:22:14.526314     8 dpu_runner.cpp:155] [UNILOG][Check Failed: handle != __null][/group/dphi_software/software/workspace/huizhang/vitis_ai_library_pack/tmp/edge/src/vart/runner/src/dpu_runner.cpp:155][VART_RUNNER_CONSTRUCTION_FAIL][Cannot create runner] cannot open library! lib=libvart-dpu-runner.so, error=libvart-dpu-runner.so: cannot open shared object file: No such file or directory

```

I didn't include this file in my distro because it is unversioned.

paroque28, 2020-08-13:
```
ln -s /usr/lib/libvart-dpu-runner.so.1 /usr/lib/libvart-dpu-runner.so
```

This hides the issue, but Yocto will complain:
```
QA issue  rdepends on app-dev [dev-deps]
```

qianglin-xlnx, 2020-08-13:
Hi @paroque28 
It looks like you're creating a platform system. You can refer to the following.
https://www.xilinx.com/html_docs/vitis_ai/1_2/gjp1592545687662.html 
For the issue you mentioned, you can add `IMAGE_INSTALL_append` = " vitis-ai-library-dev"` in the project-spec/meta-user/conf/petalinuxbsp.conf and try it again. Thank you.

qianglin-xlnx, 2022-02-16:
Hi @paroque28 

Since we haven't received your reply for a long time, we assume you have solved this issue and I'm going to close it. If you still have any questions, please feel free to reopen it. Thank you very much.

closed 2022-02-16 by qianglin-xlnx