Xilinx/Vitis-AI #146: KERNEL PANIC on TRD Vivado default petalinux config

jalezeta, 2020-09-03:
I am trying to boot the image generated through the default BSP provided on the DPU_TRD project. I am following the steps indicated in [this Makefile](https://github.com/Xilinx/Vitis-AI/tree/master/DPU-TRD/prj/Vivado). The problem is that I am getting an error message through the UART terminal while booting the Petalinux Image. This is the error message I am getting:

```
Starting kernel ...

[   29.513196] Kernel panic - not syncing: No working init found.  Try passing init= option to kernel. See Linux Documentation/admin-guide/init.rst for guidance.
[   29.527352] CPU: 1 PID: 1 Comm: swapper/0 Not tainted 5.4.0-xilinx-v2020.1 #1
[   29.534475] Hardware name: ZynqMP ZCU102 Rev1.0 (DT)
[   29.539423] Call trace:
[   29.541860]  dump_backtrace+0x0/0x148
[   29.545509]  show_stack+0x24/0x30
[   29.548810]  dump_stack+0xac/0xd0
[   29.552115]  panic+0x148/0x314
[   29.555155]  kernel_init+0xf4/0x104
[   29.558634]  ret_from_fork+0x10/0x1c
[   29.562194] SMP: stopping secondary CPUs
[   29.566110] Kernel Offset: disabled
[   29.569587] CPU features: 0x0002,20002004
[   29.573578] Memory Limit: none
[   29.576621] ---[ end Kernel panic - not syncing: No working init found.  Try passing init= option to kernel. See Linux Documentation/admin-guide/init.rst for guidance. ]---
```

andrew0917, 2020-09-04:
Hi jalezeta, As for the error log of your boot the PetaLinux Image, the fatel error is: **not syncing: No working init found**.This error usually caused by the flowing aspects:
1. rootfs.tar.gz without init executable file
2. rootfs.tar.gz isn't burnd into SD card ext4 partion correctly
3. u-boot bootargs setting is not match the the partition of SD card ext4
4. ext4 is not enable in linux kernel configuration
5. SD card is physically damaged

In my experience, I doubt the 2 item first ,Could you please access your burned SD card via Linux OS and verify if the ext4 partion is exist?  if there are /bin,/etc/,/var,/tmp..... under the ext4 partition?

jalezeta, 2020-09-08:
Thanks @andrew0917! I realized that the ext4 partition wan not generated properly even though I was able to see the rootfs partition with /bin,/etc...

I had to use the dd command and mount the rootfs using .ext4 file format instead of the .tar for some reason and that was causing the problem. Anyway, thank you very much for your support!

closed 2020-09-08 by jalezeta