Xilinx/Vitis-AI #147: undefined reference to dpuGetInputTensorHeight

kevinkit, 2020-09-08:
I am trying to get the segmentation example to work on a custom board. I got the ResNet50 example to work just fine, and now I am trying it with this example : https://github.com/Xilinx/Vitis-AI/tree/master/mpsoc/vitis_ai_dnndk_samples/segmentation/src 

However, the dputils.h and dputils.cpp only provide three functions, and the function dpuGetInputTensorHeight is not included in those. 

Is there any other location where this functions are defined, so I can add them as a source ?

qianglin-xlnx, 2020-09-09:
Hi @kevinkit 
The code implementation for dpuGetInputTensorHeight() is included in Vitis AI DNNDK runtime. You can find it in https://github.com/Xilinx/Vitis-AI/blob/v1.2.1/mpsoc/vitis_ai_dnndk_src/n2cube/src/dpu.cpp 

Also, we have provided a compiled package of Vitis AI DNNDK runtime. The download link is shown below.
https://www.xilinx.com/bin/public/openDownload?filename=vitis-ai_v1.2_dnndk.tar.gz
You can also call the APIs by linking to the DNNDK runtime libraries and including the n2cube.h head file.

kevinkit, 2020-10-05:
Hey Thanks for your answer,

Is it possible to install the VART in petalinux after it was compiled or do I need to start from the beginning and create the petalinux again?

qianglin-xlnx, 2020-10-09:
> 
> 
> Hey Thanks for your answer,
> 
> Is it possible to install the VART in petalinux after it was compiled or do I need to start from the beginning and create the petalinux again?

Hi @kevinkit 
You can refer https://www.xilinx.com/html_docs/vitis_ai/1_2/gjp1592545687662.html to install the VART in your custom board.

kevinkit, 2020-10-09:
Thanks, as you see in #167 - I was faced with some other issues

closed 2020-10-09 by kevinkit