Xilinx/Vitis-AI #148: Reference for Vitis AI Quantization

FrankXinqi, 2020-09-09:
Could I ask for the reference publication for quantization in Vitis AI? This is not mentioned in Vitis AI User Guide.

If possible, the reference for pruning as well.

Many thanks.

qianglin-xlnx, 2020-09-14:
Hi @FrankXinqi 
I'm afraid there is no reference for quantization algorithm. It's recommended that you can refer to the following 2 documents to use the quantization tools.
UG1414: https://www.xilinx.com/html_docs/vitis_ai/1_2/uim1570695919827.html
UG1333: https://www.xilinx.com/support/documentation/sw_manuals/vitis_ai/1_2/ug1333-ai-optimizer.pdf

FrankXinqi, 2020-09-14:
Many thanks @qianglin-xlnx 
I was aware of the two documents. The reason I am asking for that is that VitisAI quantization does not perform desired result to reduce model size in a lot. 

If no refernce can be provided, please refer to #150. I am wondering if the algorithm only quantize FC layers, rather than CNN and FC layers.

FrankXinqi, 2020-09-15:
Thanks for the information. If there are updates about quantization algorithms used in Vitis AI, please let me know. 

I close this issue now.

closed 2020-09-15 by FrankXinqi