Xilinx/Vitis-AI #149: Zynq7000 series  dexplorer error

zangii, 2020-09-10:
I made a microzed pfm(vitis platform) refer by Vitis_Embedded_Platform_Source git.
This pfm is good working on Vitis vadd & helloworld example .
So, I working on Vitis-AI 1.2.1 with this pfm.
However, dpu does not working.

Is it not possible  zynq7000 in Vitis-AI?
In the pfm made of ultra96 the dpu is good working.

Error Log : 

root@microzed-2020_1:/mnt/sd-mmcblk0p1# xbutil program -p dpu.xclbin
INFO: Found total 1 card(s), 1 are usable
[drm] Pid 569 opened device
[drm] zocl_xclbin_read_axlf The XCLBIN already loaded, uuid: d081fd94-9c3a-4767-b61c-98d80ac4903d. Don't need to reload.
[drm] zocl_xclbin_read_axlf d081fd94-9c3a-4767-b61c-98d80ac4903d ret: 0.
[drm] -> Hold xclbin D081FD94-9C3A-4767-B61C-98D80AC4903D, from ref=0
[drm] <- Hold xclbin D081FD94-9C3A-4767-B61C-98D80AC4903D, to ref=1
[drm] Reconfiguration not supported
[drm] -> Release xclbin D081FD94-9C3A-4767-B61C-98D80AC4903D, from ref=1
[drm] now xclbin can be changed
INFO: xbutil program succeeded.
[drm] <- Release xclbin D081FD94-9C3A-4767-B61C-98D80AC4903D, to ref=0
[drm] Pid 569 closed device
root@microzed-2020_1:/mnt/sd-mmcblk0p1# xbutil query
INFO: Found total 1 card(s), 1 are usable
~~~~~~~~~~~~~~~~~~~~~~~~~~[drm] Pid 570 opened device
~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
System C[drm] Pid 570 opened device
onfiguration
OS name:        Linux
Release:        5.4.0-xilinx-v2020.1
Version:        #1 SMP PREEMPT [drm] Pid 570 closed device
Thu Sep 10 00:18:40 UTC 2020
Machine:        armv7l
Glibc:          2.30
Dis[drm] Pid 570 closed device
tribution:      N/A
Now:            Thu Sep 10 01:01:11 2020
~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
XRT Information
Version:        2.6.0
Git Hash:       2d6bfe4ce91051d4e5b499d38fc493586dd4859a
Git Branch:     2020.1
Build Date:     2020-09-10 00:21:43
ZOCL:           2018.2.1
~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
Shell                           FPGA                            IDCode
xilinx_microzed_base_202010_1   N/A                             N/A
Vendor          Device          SubDevice       SubVendor
0x10ee          N/A             N/A             N/A
DDR size        DDR count       Clock0          Clock1          Clock2
0 Byte          1               100             0               0
~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
Memory Status
     Tag         Type        Temp(C)  Size    Mem Usage       BO count
[ 0] HP0         MEM_DRAM             1 GB    0 Byte          24
[ 1] HP1         MEM_DRAM             1 GB    0 Byte          24
[ 2] HP2         MEM_DRAM             1 GB    0 Byte          24
[ 3] HP3         **UNUSED**           0 Byte  0 Byte          24
~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
Streams
     Tag         Flow ID  Route ID Status   Total (B/#)     Pending (B/#)
~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
Xclbin UUID
d081fd94-9c3a-4767-b61c-98d80ac4903d
~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
Compute Unit Status
CU[ 0]: DPUCZDX8G:DPUCZDX8G_1           @0x80000000        (IDLE)
~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
INFO: xbutil query succeeded.
root@microzed-2020_1:/mnt/sd-mmcblk0p1# dexplorer -w
[drm] Pid 571 opened device
[drm] Pid 571 closed device
[drm] Pid 571 opened device
[drm] zocl_xclbin_read_axlf The XCLBIN already loaded, uuid: d081fd94-9c3a-4767-b61c-98d80ac4903d. Don't need to reload.
[drm] zocl_xclbin_read_axlf d081fd94-9c3a-4767-b61c-98d80ac4903d ret: 0.
[drm] -> Hold xclbin D081FD94-9C3A-4767-B61C-98D80AC4903D, from ref=0
[drm] <- Hold xclbin D081FD94-9C3A-4767-B61C-98D80AC4903D, to ref=1
[drm] Reconfiguration not supported
[drm] -> Release xclbin D081FD94-9C3A-4767-B61C-98D80AC4903D, from ref=1
[drm] now xclbin can be changed
[drm] <- Release xclbin D081FD94-9C3A-4767-B61C-98D80AC4903D, to ref=0
[drm] Pid 571 opened device
[drm] CU can only be initialized once.
XRT build version: 2.6.0
Build hash: 2d6bfe4ce91051d4e5b499d38fc493586dd485[drm] Pid 571 closed device
9a
Build date: 2020-09-10 00:21:43
Git branch: 2020.1
PID: 57[drm] Pid 571 closed device
1
UID: 0
[Thu Sep 10 01:01:51 2020]
HOST: microzed-2020_1
EXE: /usr/bin/dexplorer
[XRT] ERROR: getVirtAddressOfApture: Could not found the mapped address. Check if XCLBIN is loaded.
[XRT] ERROR: xclRead: Invalid offset.
[DNNDK_XRT] Error when reading regs, phy_addr:0x0000000000000000, count: -2147483616, return bytes: -1
[XRT] ERROR: getVirtAddressOfApture: Could not found the mapped address. Check if XCLBIN is loaded.
[XRT] ERROR: xclRead: Invalid offset.
[DNNDK_XRT] Error when reading regs, phy_addr:0x0000000000000000, count: -2147483392, return bytes: -1
[DNNDK] Invalid hp width '0' found in DPU signature.

Error: fail to open DPU device: please check if DPU Driver works as expected.

qianglin-xlnx, 2020-09-28:
hi @zangii 
We have confirmed that there is bug for Zynq7000 in Vitis Flow. We have fixed it and it will be updated until next release.
For Zynq7000, you can use Vivado flow and try it again. We also confirmed that it's normal in Vivado flow.
Thank you for your feedback.

sl33k, 2020-11-16:
hi @qianglin-xlnx,

is there any way you could provide a work-around for the issue until the next version is released?
Or at the least, could you specify the bug exactly so one can develop such a work-around?

Thanks!

x3chen, 2020-11-16:
Hi @sl33k ,
1) dexplorer problem on ZYNQ-7000 is caused by function dpu_aol_read_regs in vitis_ai_dnndk_src.
2) we also found bugs in DPU IPCORE, so about VITIS AI 1.2 on ZYNQ-7000, the vivado flow is recommended.

lllkiller-jlll, 2020-11-20:
@qianglin-xlnx @x3chen 

Is there any tutorial for Zynq7000 using Vivado?

x3chen, 2020-11-22:
Vivado flow is basically the same on zynq and mpsoc. 
It includes DPU IPCORE integration, DTS modification, Driver compilation and runtime installation.
Refer to the the following link:
  https://github.com/Xilinx/Vitis-AI/tree/master/DPU-TRD/prj/Vivado
  https://github.com/Xilinx/Vitis-AI/tree/master/mpsoc/vitis_ai_dnndk_src

sl33k, 2020-11-23:
@x3chen 
The tutorial you linked does not specify exactly what DTS modifcations are required. A few more questions:

1. I assume a zynqclmm_drm node is required as with the vitis flow?
2. The Vitis flow includes an AXI interrupt controller, can you specify if that is actually required, because the default vivado flow does not generate such a controller.
3. Does the driver compilation use XRT or the default linux driver? How does XRT find the address of the DPU?

Thanks!

x3chen, 2020-11-23:
For vivado flow, 
1. zynqclmm_drm node isn't required. please refer to the following BSP.
    https://github.com/Xilinx/Vitis-AI/tree/master/DPU-TRD/prj/Vivado/dpu_petalinux_bsp
2. intc isn't required too. u can directly connect DPU interrupt to ps
3. DPU driver link
    https://github.com/Xilinx/Vitis-AI/tree/master/mpsoc/vitis_ai_dnndk_src/driver

sl33k, 2020-11-23:
@x3chen 
Thanks for all the help so far.
What I mean by question 3. is if the address set in the https://github.com/Xilinx/Vitis-AI/blob/b26c871ccdc6bf361c1454dd919c9af842e1dee0/mpsoc/vitis_ai_dnndk_src/driver/Makefile#L25 needs to match the address in the address editor in Vivado.
The address is autogenerated to ```0x4000_0000``` by vivado. The Makefile on the other hand specifies ```0x4FF0_0000```.

x3chen, 2020-11-23:
1) vivado flow isn't based on XRT. please choose linux for runtime compilation
https://github.com/Xilinx/Vitis-AI/blob/master/mpsoc/vitis_ai_dnndk_src/Makefile.cfg#L8-12
2) DPU base address is read from dts node, NOT SIG_BASE_ADDR
https://github.com/Xilinx/Vitis-AI/blob/master/mpsoc/vitis_ai_dnndk_src/driver/dpucore.c#L748-754

qianglin-xlnx, 2022-02-18:
Hi @zangii 
Since we haven't received your reply for a long time, we assume you have solved this issue and I'm going to close it. If you still have any questions, please feel free to reopen it. Thank you very much.

closed 2022-02-18 by qianglin-xlnx