Xilinx/Vitis-AI #150: Vitis AI Quantization does NOT reduce Tensorflow model size

FrankXinqi, 2020-09-12:
I use Vitis AI to quantize my tensorflow model (CNN+FC). However, it only has 0.1 MB reduce from 43.2MB to 43.1MB.
I have also checked with other people who use it, and they have the same result.

I am wondering if the algorithm only qunatize FC layers. Hope to get developers support!

FrankXinqi, 2020-09-14:
To provide one complete information about sizes:
original keras model: 39.7MB
converted tensorflow model: 39.6MB
quantize_eval_model.pb: 39.5MB
deploy_model.pb: 39.5MB

qianglin-xlnx, 2020-09-14:
Hi @FrankXinqi 
The quantizer will quantize both Conv and FC layers, but the values in the output models are still float values, thus you will not see the size reduce. The quantizer will add some quantize operations in the Conv and FC layers to simulate the quantize error, which enables the post training quantization and quantize-aware training. In the compiler, the float values will be finally converted to INT8, and the size of the compiled model will reduce.

FrankXinqi, 2020-09-14:
Thanks @qianglin-xlnx. I think the quantized model still has float32 and other points are important, please consider add them in the official documents. I also suggest you to add related reference if any xilinx developers published related papers. Please ignore if no publication.

I get the point that after compile the model will be reduced finally, and the quantize_eval_model.pb are added with some quantize operations to simulate the quantize error.

However, I the compiled model is not comparable with the pb model, so how I can know the reduced size of my model? In addition, I do not get the idea why xilinx simulate the quantization error, rather than directly providing a quantized model for us.

qianglin-xlnx, 2020-09-14:
@FrankXinqi 
Thank you for your advice.
After you use the AI compiler to compile the model, you will finally get the compiled model and know the reduced size of the model. 
Xilinx also provided the compiled models for different platforms. 
For zcu102, https://www.xilinx.com/bin/public/openDownload?filename=xilinx_model_zoo_zcu102-1.2.0-1.aarch64.rpm 
For zcu104, https://www.xilinx.com/bin/public/openDownload?filename=xilinx_model_zoo_zcu104-1.2.0-1.aarch64.rpm

For alveo card, find the compiled models in the following link
https://github.com/Xilinx/Vitis-AI/tree/master/Vitis-AI-Library#setting-up-the-host-for-u50u50lvu280

FrankXinqi, 2020-09-14:
Thanks @qianglin-xlnx 
I can get the compile model, but with the compile model is ready for deployment with different file type. In my opininon, the comparison should make between the same file types.

What I need is to calculate the performance after quantization without a board. It seems quantize_eval_model.pb is able to do the job, by simulating quantization error.  Am I correct?  In addition, why Vitis AI does this complicated process to simulate the error, rather than provide the quantized model directly.

qianglin-xlnx, 2020-09-14:
Hi @FrankXinqi 
I think it's hard to calculate the performance after quantization without running it on the board. The performance of the model depends on many factors, including pre-processing and post-processing, DPU IP, hardware platform, memory, IO, etc. It is recommended to actually run the model to obtain accurate performance.
Maybe, you can refer to the published models for specific information and performance data, then get a rough performance evaluation of your model, as shown in the following.
[image]

closed 2020-10-23 by FrankXinqi