Xilinx/Vitis-AI #151: Vitis AI Compiler has no extra option "quant_cfgfile".

lllkiller-jlll, 2020-09-17:
Hi.

I'm trying to use own model on Alveo U250 accelerator card.

I saw (https://github.com/Xilinx/Vitis-AI/blob/master/alveo/examples/caffe/run.py) and I made deploy.prototxt, deploy.caffemodel, and quantize_info.txt successfully.

Finally, I input vai_c_caffe command but I faced the error message below:

`
**************************************************
* VITIS_AI Compilation - Xilinx Inc.
**************************************************
Traceback (most recent call last):
  File "/opt/vitis_ai/conda/envs/vitis-ai-caffe/bin/vai_c_caffe", line 196, in <module>
    compiler = VAI_Caffe_Frontend(args)
  File "/opt/vitis_ai/conda/envs/vitis-ai-caffe/bin/vai_c_caffe", line 93, in __init__
    u = eval(args.options) if args.options else {}
  File "<string>", line 1, in <module>
NameError: name 'quant_cfgfile' is not defined
`

My command was :
`
vai_c_caffe --prototxt quantize/deploy.prototxt --caffemodel quantize/deploy.caffemodel --output_dir compile/ --net_name tmp --arch /opt/vitis_ai/compiler/arch/dpuv1/ALVEO/ALVEO.json --options {"quant_cfgfile":"quantize/quantize_info.txt"}
`

What did I miss?

Thanks.

admin-myte, 2020-09-17:
Can you try following:
vai_c_caffe --prototxt quantize/deploy.prototxt --caffemodel quantize/deploy.caffemodel --output_dir compile/ --net_name tmp --arch /opt/vitis_ai/compiler/arch/DPUCADX8G/ALVEO/arch.json --options "{'quant_cfgfile':'quantize/quantize_info.txt'}"

Are you using the latest version of run.py?

sumitn-xilinx, 2020-09-17:
Can you try following:
vai_c_caffe --prototxt quantize/deploy.prototxt --caffemodel quantize/deploy.caffemodel --output_dir compile/ --net_name tmp --arch /opt/vitis_ai/compiler/arch/DPUCADX8G/ALVEO/arch.json --options "{'quant_cfgfile':'quantize/quantize_info.txt'}"

Are you using the latest version of run.py?

lllkiller-jlll, 2020-09-18:
Hi sumitn.

I succeed convert custom model for DPUv1.

Thank you very much!

I'll close this issue.

closed 2020-09-18 by lllkiller-jlll