Xilinx/Vitis-AI #152: VITIS-AI docker gpu error

nevilshah235, 2020-09-17:
the gpu docker image has been successfully installed, but i am getting this error when I run the command
**sudo ./docker_run.sh xilinx/vitis-ai-gpu:latest**

docker: Error response from daemon: OCI runtime create failed: container_linux.go:349: starting container process caused "process_linux.go:449: container init caused \"process_linux.go:432: running prestart hook 0 caused \\\"error running hook: exit status 1, stdout: , stderr: nvidia-container-cli: initialization error: driver error: failed to process request\\\\n\\\"\"": unknown.

hanxue, 2020-09-23:
Hi @nevilshah235 ,

The nvidia-docker2 package has been updated to use `--gpus all` instead of `--runtime=nvidia`, see https://docs.nvidia.com/datacenter/cloud-native/container-toolkit/install-guide.html#docker 

Please can you edit `docker_run.sh` and change `--runtime=nvidia` to `--gpus all`?

We will be updating the docker_run.sh script and documentation in the next release.

nevilshah235, 2020-09-24:
Hi @hanxue 

Thank you very much. It worked perfectly!

closed 2020-09-24 by nevilshah235