Xilinx/Vitis-AI #154: vai_c_tensorflow compile error with DPUCADX8G

World-Worst-Detector, 2020-09-21:
I am trying to compile the quantized unet model with vai_c_tensorflow. The docker container is built from master branch with GPU support. Inside vitis-ai-tensorflow conda, I used the following command line:

vai_c_tensorflow --frozen_pb ./quantize/unet3/deploy_model.pb --arch /opt/vitis_ai/compiler/arch/DPUCADX8G/ALVEO/arch.json --output_dir ./compile  --net_name unet3 --options "{'quant_cfgfile':'./compile/unet3_fix.txt'}

The output is like following:

Traceback (most recent call last):
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow/bin/vai_c_tensorflow", line 203, in <module>
    compiler.compile()
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow/bin/vai_c_tensorflow", line 177, in compile
    os.rename(DIR+weight,    DIR+'weights.h5'    )
FileNotFoundError: [Errno 2] No such file or directory: './compile/deploy_model.pb_data.h5' -> './compile/weights.h5'

Do you have any suggestions?

Thank you for your help.

lllkiller-jlll, 2020-09-25:
I met same problem.

Please let me know solution.

katherineyun, 2020-09-29:
I also encountered the same issue. Hope someone can look into this soon.

fjzzlhc, 2020-11-17:
I met the problem before. The error disappeared affter I added the option " --quant_info" with no other parameter. You can have a try.

World-Worst-Detector, 2020-11-23:
> I met the problem before. The error disappeared affter I added the option " --quant_info" with no other parameter. You can have a try.

Thank you for sharing the information. With '--quant_info', the error has gone. But, it can not generate the Weight.h5, compiler.json and quantizer.json.