Xilinx/Vitis-AI #156: cf_SPnet_aichallenger_224_128_0.54G_1.2 pre-quantized model does not compile

dsp-avt, 2020-09-21:
The cf_SPnet_aichallenger_224_128_0.54G_1.2 pre-quantized model available in quantized/Edge directory does not compile.

Model: https://www.xilinx.com/bin/public/openDownload?filename=cf_SPnet_aichallenger_224_128_0.54G_1.2.zip

Compilation command executed in model directory:

```bash
vai_c_caffe \
--prototxt quantized/Edge/deploy.prototxt \
--caffemodel quantized/Edge/deploy.caffemodel \
--arch /opt/vitis_ai/compiler/arch/DPUCZDX8G/ZCU104/arch.json \
--output vai_c_output \
--net_name sp_net
```

Error:
```bash
[VAI_C][Fatal] Check failed for condition [layer_id < model.layer_size()] in [/home/xbuild/conda-bld/dnnc_1592904456005/work/dnnc_impl/parser/caffe/caffeparser.cc:185] :Failed to find parameter for [fc_visible] in caffemodel, make sure name in prototxt is consistent with that in caffemodel.
```

This issue is not present with the Vitis-AI v1.1 model release - https://www.xilinx.com/bin/public/openDownload?filename=cf_SPnet_aichallenger_224_128_1.1.zip

qianglin-xlnx, 2020-10-15:
Hi @dsp-avt 
Thank you for your feedback. Please find the updated deploy.caffemodel in the attachment and try it again. 
[sp_net.tar.gz](https://github.com/Xilinx/Vitis-AI/files/5382450/sp_net.tar.gz)

dsp-avt, 2020-10-21:
@qianglin-xlnx - The updated model compiled successfully.  Thank you.

closed 2020-10-21 by dsp-avt