Xilinx/Vitis-AI #157: docker: Error response from daemon: Unknown runtime specified nvidia

kevinkit, 2020-09-22:
I had this error when trying to run the gpu container. I could resolve it by changing the 

`--runtime=nvida `to `--gpus all `

as discussed here:

https://github.com/NVIDIA/nvidia-docker/issues/838 

I am not sure if this is related to the docker image and if should be changed in upcoming releases

hanxue, 2020-09-23:
Hi @kevinkit ,

The nvidia-docker2 package has been updated to use `--gpus all` instead of `--runtime=nvidia`, see https://docs.nvidia.com/datacenter/cloud-native/container-toolkit/install-guide.html#docker 

We will be updating the docker_run.sh script and documentation in the next release.

serser, 2022-01-20:
For me, I need to install nvidia runtime by https://stackoverflow.com/questions/52865988/nvidia-docker-unknown-runtime-specified-nvidia

closed 2022-03-03 by qianglin-xlnx