Xilinx/Vitis-AI #158: Support for Zynq7000(such as ZedBoard)

sunshine411, 2020-09-25:
Hi.
Is it possible to use Vitis AI on zynq7000?
If it works, what is the development process (take zedboard as an example)
How do I need to build the operating environment of the board end, do you have a system image that can run.

kevinkit, 2020-09-29:
It is supported but it will be a hard way since the documentation is scattered xilinx style all around the world wide web. If you want to get it up quick and runnig a ZCU102/104 Board will help you a lot. Furthermore, I read that several features are not supported by the Zynq7000. To quote from  @wilderfield  #138 :

"While not prepackaged, with enough elbow grease, anyone can run Vitis AI apps on any Zynq UltraScale+ board. One needs to build the base platform with Vivado, build the firmware and root file system with PetaLinux, and insert the DPU accelerator + build bitstream with Vitis."

If you are familiar with Vivado/VItis, have a little experience in Embedded Linux and Deep Learning you will get this up and running with some time investement.  However, having experience in all required fields is absolutely necessary and seldom found in one person. 

On the plus side @andyluo7 has written in the same thread that the will have detailed tutorials on how to port VITIS Ai to custom platfroms "soon" - but has not answered any tags since. 

Furthermore, I really liked this tutorial: https://github.com/hill19850213/vitis_ai_custom_platform_v1.2.1

hopesala, 2020-10-15:
ZCU106 support?

kevinkit, 2020-10-15:
@hopesala there are a lot of other issues that are already discussing zcu106 support in detail, please use the github search function for issues: #38 #138

qianglin-xlnx, 2022-02-20:
Hi @sunshine411 
Since we haven't received your reply for a long time, we assume you have solved this issue and I'm going to close it. If you still have any questions, please feel free to reopen it. Thank you very much.

closed 2022-02-20 by qianglin-xlnx