Xilinx/Vitis-AI #162: DPUCZDX8G DPU configuration

kerocc, 2020-10-02:
Hi,

Is the DPUCZDX8G DPU configuration described somewhere? This is the default DPU configuration available in Vitis AI for ZCU102 boards. **Especially, I would like to know the value of the "channel_parallel" variable.**

Maybe I missed the information in the Vitis AI User Guide... In UG1414 the channel_parallel for DPUCAHX8H is said to be 16, but I cannot find something about DPUCZDX8G.

Thanks a lot.

qianglin-xlnx, 2020-10-09:
@kerocc 
You can find the DPU configuration in PG338.  Please download it in the following link.
https://www.xilinx.com/support/documentation/ip_documentation/dpu/v3_2/pg338-dpu.pdf
Also, you can find DPU configuration in 
https://github.com/Xilinx/Vitis-AI/tree/master/DPU-TRD

kevinkit, 2020-10-21:
[image]

kerocc, 2020-10-21:
Thanks for your replies. I had these tables in mind, they are really useful to configure our own DPU.

But here I was wondering what is the architecture provided with the ZCU102 image that we find in the VART Tutorials.
Is it B4096/3 cores? This is the information I was looking for. Is it given somewhere?

Thanks.

kevinkit, 2020-10-21:
I am not quiture sure but you can check with `dexplorer -w` on your target board how many dpu cores are present. Sadly I do not have a zcu102, but I assume it is the common used TRD from xilinx which includes 2 DPU Cores. 

When you have more DPU cores you do not automatically have the whole throughput of 2 DPU cores, only a single DPU is utilized when you do not write the appropiate code to use both. So e.g. using the first DPU for every even and the second for every uneven frame

qianglin-xlnx, 2020-10-21:
@kerocc 
There are 3 * B4096F DPU core @300Mhz integrated in the official ZCU102 image.

closed 2020-10-21 by kerocc