Xilinx/Vitis-AI #163: caffe-xilinx runtest error ghosting caffe::SyncedMemory::mutable_cpu_data()

vxgu86, 2020-10-04:
for days I have been trying to dig this problem without results.
by the way. I tried the BVLC caffe without any error.

[----------] 8 tests from RMSPropSolverTest/1, where TypeParam = caffe::CPUDevice<double>
[ RUN      ] RMSPropSolverTest/1.TestLeastSquaresUpdateWithEverythingAccum
*** Aborted at 1601787254 (unix time) try "date -d @1601787254" if you are using GNU date ***
PC: @     0x7f355884e27b caffe::SyncedMemory::mutable_cpu_data()
*** SIGSEGV (@0x18) received by PID 15321 (TID 0x7f355f6d4280) from PID 24; stack trace: ***
    @     0x7f355a13f390 (unknown)
    @     0x7f355884e27b caffe::SyncedMemory::mutable_cpu_data()
    @     0x7f3558798481 caffe::Blob<>::CopyFrom()
    @           0x7ee3ad caffe::GradientBasedSolverTest<>::CheckAccumulation()
    @           0x91e333 testing::internal::HandleExceptionsInMethodIfSupported<>()
    @           0x917caa testing::Test::Run()
    @           0x917df8 testing::TestInfo::Run()
    @           0x917ed5 testing::TestCase::Run()
    @           0x91818f testing::internal::UnitTestImpl::RunAllTests()
    @           0x9184c3 testing::UnitTest::Run()
    @           0x47258b main
    @     0x7f3557bc5840 __libc_start_main
    @           0x47a769 _start
    @                0x0 (unknown)
Makefile:560: recipe for target 'runtest' failed
make: *** [runtest] Segmentation fault (core dumped)

lji72, 2020-10-09:
same error likes https://github.com/Xilinx/Vitis-AI/issues/164.

qianglin-xlnx, 2022-02-17:
Hi @vxgu86  Please refer to #164 .

closed 2022-02-17 by qianglin-xlnx