Xilinx/Vitis-AI #164: file missing when compile caffe_xilinx1.1 with make test

vxgu86, 2020-10-04:
Caffe_Xilinx1.1$ make test
CXX src/caffe/test/test_conv_bn_fixed_layer.cpp
src/caffe/test/test_conv_bn_fixed_layer.cpp:9:54: fatal error: caffe/layers/cudnn_conv_bn_fixed_layer.hpp: No such file or directory
compilation terminated.
Makefile:600: recipe for target '.build_release/src/caffe/test/test_conv_bn_fixed_layer.o' failed
make: *** [.build_release/src/caffe/test/test_conv_bn_fixed_layer.o] Error 1

**all in all, do we have to run "make test/make runtest" or we just have to run the "make all"**

vxgu86, 2020-10-04:
I mean, if we do not do these tests, will the accuracy be good?

lji72, 2020-10-09:
Hi, @vxgu86 , user just needs run the "make all". 
We add some new layers to caffe-xilinx without test files.

closed 2020-10-10 by vxgu86