Xilinx/Vitis-AI #165: What's the point of dlet?

kevinkit, 2020-10-05:
When having a .hww file and using dlet to generate a .dcf file it will simply put the descriptions from the hwh file from the DPU into the .dcf file. Furthermore, some binary header is added. 

I was wondering what the point is? Why can't the hwh file be used in the arch.json for generating the .elf file - is it a structural decision or is there more behind this?

Mookel, 2020-10-09:
> When having a .hww file and using dlet to generate a .dcf file it will simply put the descriptions from the hwh file from the DPU into the .dcf file. Furthermore, some binary header is added.

Hi, @kevinkit it's more than just simply put the descriptions from the hwh file into the dcf file. We use dcf with different considerations:
1. The format and content of HWH file may change in the future. With an indepedent dcf file we can remove the unnecessary dependency between the compiler and HWH files, which means we only need to upgrade DLet if the format/content of HWH change;
2. Use an indepedent dcf file, our compiler can also support more configuration file types other than the HWH file without any changes and just needs to add a new parser for our DLet tool;
3. The DLet can originize the data in HWH into structual representation more friendly to our compiler, thus making compiler's life easier:-).

Hope it helps instead of confusion, thanks very much.

kevinkit, 2020-10-09:
Thanks this helps a lot and gave a better understanding of the principles

closed 2020-10-09 by kevinkit