Xilinx/Vitis-AI #166: 'vai_q_tensorflow quantize' complains about 'bad shape'

RSPwFPGAs, 2020-10-05:
I am trying to compile a pre-trained model with Vitis AI 1.2, following the example in [https://github.com/Xilinx/DPU-PYNQ/blob/master/host/train_mnist_model.ipynb](url). I got stopped at the Quantization step.
The pre-trained model can be properly loaded and get correct inference result. So it must be one of the Vitis AI specific steps before quantization that is  wrong, or maybe there is unsupported ops in the model.

I attached the jupyter notebook here, which contains the detailed error report and model nodes. Thank you for looking into it!
[poly_yolo_pynq.ipynb.zip](https://github.com/Xilinx/Vitis-AI/files/5328049/poly_yolo_pynq.ipynb.zip)

kevinkit, 2020-10-05:
i faced the same problem, sometimes it is due to the wrong output layer. 

Another option is that conv2d->relu->bn must be written as conv2d->bn->relu - I have no idea why and this is a major bug.

RSPwFPGAs, 2020-10-09:
Thank you @kevinkit, for you advice! I have tried different convolution layers for the output node, but they all failed with the same error.

kevinkit, 2020-10-09:
Did you check the conv2d -> batchnorm -> relu order that has to be used?

xieyi4650, 2020-10-09:
@RSPwFPGAs   according to error message, it is caused by the wrong input tensor dim (beta) of BN node.  The batch norm node maybe not the typical one.  So could you please check your BN pattern which means the input of BN node and its index. Or you can share your BN node sub-graph. 
You can also share your graph directly.

kevinkit, 2020-10-09:
Thanks to @xieyi4650 comment I am now pretty sure that you are facing this issue:

https://forums.xilinx.com/t5/AI-and-Vitis-AI/quot-Beta-input-to-batch-norm-has-bad-shape-quot-Error-while/td-p/1031063

So Basically reallocate `Conv2D -> Relu` -> bn to `Conv2D -> bn -> relu`

This happens due the quantization optimization steps which "integrates" the batchnorm into the conv layer. Thus the relu must come AFTER the batch norm layer

xieyi4650, 2020-10-09:
Thanks @kevinkit .   So you mean that `Conv2D -> Relu -> bn` will trigger the bug? After checking code, I found that the error is not thrown during integrate the batchnor into conv2D layer.  I guess this is caused by different input order of BN node.  Could please share the model with mock weights,  so we can analysis the situation? Thanks

kevinkit, 2020-10-09:
Sadly, I do not have the option to share the model etc. atm (home office ;) ) - however, one can easlily reproduce the bug by using the keras application module and use the "MobileNet_v2" oder "MobileNet" here

xieyi4650, 2020-10-09:
@kevinkit  OK.  To reproduce the the bug,  can you tell your tensorflow version and keras version

kevinkit, 2020-10-09:
TF: 1.15.0
keras: 2.3.1

RSPwFPGAs, 2020-10-10:
@xieyi4650 I uploaded the whole project here. My environment is xilinx/vitis-ai-cpu:latest and vitis-ai-tensorflow with matplotlib keras==2.2.5 installed.
[poly_yolo_dpu.1010.zip](https://github.com/Xilinx/Vitis-AI/files/5358558/poly_yolo_dpu.1010.zip)
[models/poly_yolo.h5](https://gitlab.com/irafm-ai/poly-yolo/-/blob/master/poly_yolo/models/poly_yolo.h5)

RSPwFPGAs, 2020-10-10:
@kevinkit I captured the first layer of the compute graph. As can be seen, the order is conv2d -> batchnorm -> relu, with some variations. 
[image]

xieyi4650, 2020-10-10:
Hi @RSPwFPGAs  I suggest you try to remove the nodes associated with `switch` then run quantize.  For the inference graph, typical structure  of `conv2D -> bn -> Relu` is shown below.

[image]

RSPwFPGAs, 2020-10-10:
Hi @xieyi4650, it seems that the "Switch" and "Merge" ops are not supported in DPU or TFlite or TensorRT. After training(freezing?), they should be combined with "FuseBatchNormV3" into one op. I am trying to find a way to do that.

xieyi4650, 2020-10-10:
@RSPwFPGAs  Another question,  I found from the picture that dim of the `mean` is 0,  but in theory it should be equal to the number of channels aka 24.

RSPwFPGAs, 2020-10-10:
@xieyi4650 There are two "FusedBatchNormV3" branches in the graph, and they are later "Merged" into one. The left branch has the dim of the mean of 0, but the right branch has the dim of the mean of 24.

RSPwFPGAs, 2020-10-10:
I am looking into the model definition of [the example CNN model for MNIST](https://github.com/Xilinx/DPU-PYNQ/blob/master/host/train_mnist_model.ipynb). I wonder why there is no ops as "Switch" or "Merge" in its original pb file. Is it because of the way its model is defined?

> model = keras.models.Sequential([
    InputLayer(input_shape=(28, 28, 1), name='input_data'),
    Conv2D(32, 3, activation='relu'),
    MaxPool2D(pool_size=(2,2)),
    Flatten(),
    Dense(128, activation='relu'),
    Dense(10, activation='softmax', name='output_logits')
])
>

xieyi4650, 2020-10-10:
Hi @RSPwFPGAs  for better compatibility,  I would like to suggest that use tf.keras rather than keras.

RSPwFPGAs, 2020-10-10:
Tensorflow provides [Grappler](https://github.com/tensorflow/docs/blob/master/site/en/guide/graph_optimization.ipynb) to get a cleaner sub-graph by constant_folding and pruning.
TensorRT provides [Graph Surgeon](https://forums.developer.nvidia.com/t/tensorrt4-0-1-6-batchnormalization-support-problem/69133) to solve exactly the same BN problem.

I wonder is there an official procedure provided by Vitis AI to solve this problem?

xieyi4650, 2020-10-10:
None at the moment.  The problem has been solved in tf.keras if I remember correctly.  So would you please try it.

RSPwFPGAs, 2020-10-10:
Maybe I should set the learning phase to "inference" before loading the model.

> #Set the learning phase to Test since the model is already trained.
>keras.backend.set_learning_phase(0)
>
>#load pretrained model
>#if you want to detect more objects, lower the score and vice versa
>trained_model = yolo.YOLO(model_path='models/poly_yolo.h5', iou=0.5, score=0.3)
>

I will give it a try when I am back in office.
[Example of set_learning_phase](https://www.programcreek.com/python/example/93710/keras.backend.set_learning_phase)

RSPwFPGAs, 2020-10-12:
It works now! The "Switch" and "Merge" ops are now removed from the computation graph.
[image]

Thank you guys, @kevinkit @xieyi4650. I learned a lot in this process!

closed 2020-10-12 by RSPwFPGAs

kevinkit, 2020-10-12:
@xieyi4650 afaik it is not solved in tf 1.15.2 (the only one which is currently supported by VITIS-AI , see #155

xieyi4650, 2020-10-12:
@RSPwFPGAs  good.  It was careless of me not to review this code carefully.  It should be setting the learning phase to "inference" before loading the model

xieyi4650, 2020-10-12:
@kevinkit  it was not  solved by graph surgeon from NVIDIA. You just need to use inference graph to freeze pb file. Like RSPwFPGAs  did.

kevinkit, 2020-10-12:
His error was solved yes, but the same error message will occur when one will use Conv2D->Relu->BN instead of Conv2D->BN->Relu - as far as I see in @RSPwFPGAs case it was the Conv2d->Relu->Bn. This issue is neither fixed with setting the learning phase to 0, nor with tf.keras - since VITIS AI only supports Tensorflow 1.15. However, nvidias graph surgeon is capable of resolving the Batchnorm issue - which is NOT the problem that @RSPwFPGAs was facing, but will result in the same error message as mentioned in this issue. Thus this is only one solution to the problem, but it won't cover all cases where this problem may occur.

kevinkit, 2020-10-12:
https://forums.xilinx.com/t5/AI-and-Vitis-AI/V3-0-190624-TF-16-04LTS-Decent-q-quantization-does-not-happen/m-p/1011348#M1984

xieyi4650, 2020-10-12:
@kevinkit  Actually for `Conv2D -> Relu -> BN` pattern, quantize tool will treat it as two seperated pattern,  `Conv2D -> Relu` and `BN`.  This will not run fold BN into Conv2D, Theoretically it won't encounter this error.  Maybe it is still not specify the running phase as inference.

kevinkit, 2020-10-12:
it will, will post snippet asap