Xilinx/Vitis-AI #167: rpm command not found

kevinkit, 2020-10-05:
I am trying to instsall VART on a custom board, however it cannot find rpm:

`bash: rpm: command not found`

or:

`sh: rpm: command not found`

It also does not work with dpkg....

qianglin-xlnx, 2020-10-09:
@kevinkit 
It seems that there are no rpm and dpkg command in the custom system.
You can unzip the vart packet in your host, then copy the vart libraries to your custom board. The following are the unzip command.

rpm2cpio libunilog-1.2.0-r<x>.aarch64.rpm | cpio -idmv
rpm2cpio libxir-1.2.0-r<x>.aarch64.rpm | cpio -idmv
rpm2cpio libtarget-factory-1.2.0-r<x>.aarch64.rpm | cpio -idmv
rpm2cpio libvart-1.2.0-r<x>.aarch64.rpm | cpio -idmv

After you unzip the vart packet, you will find the vart libaries under ./usr/lib/.

kevinkit, 2020-10-12:
Just to be clear, I will find the vart libraries **on the host machine** in **./usr/lib** and will then copy it on my **target device** in the folder **./usr/lib/**

qianglin-xlnx, 2020-12-03:
> 
> 
> Just to be clear, I will find the vart libraries **on the host machine** in **./usr/lib** and will then copy it on my **target device** in the folder **./usr/lib/**

Yes. This method works, too.

closed 2020-12-03 by qianglin-xlnx