Xilinx/Vitis-AI #168: failed with caffe-xilinx1.2 and errors of the tests with mutable_cpu_data

vxgu86, 2020-10-06:
I test the caffe-xilinx all these days, with the failed of make runtest and errors among the process,
I could not figure out what is the problemm could someone tell me?
I omitted some tests, orelse the runtest poped out with errors like this:

`
[       OK ] NetTest/0.TestLossWeightMidNet (4 ms)
[ RUN      ] NetTest/0.TestSharedWeightsResume
*** Aborted at 1601950337 (unix time) try "date -d @1601950337" if you are using GNU date ***
PC: @     0x7f0b3a56b27b caffe::SyncedMemory::mutable_cpu_data()
*** SIGSEGV (@0x18) received by PID 23790 (TID 0x7f0b413f1280) from PID 24; stack trace: ***
    @     0x7f0b3be5c390 (unknown)
    @     0x7f0b3a56b27b caffe::SyncedMemory::mutable_cpu_data()
    @     0x7f0b3a4b4941 caffe::Blob<>::CopyFrom()
    @           0x4e8616 caffe::NetTest_TestSharedWeightsResume_Test<>::TestBody()
    @           0x91e333 testing::internal::HandleExceptionsInMethodIfSupported<>()
    @           0x917caa testing::Test::Run()
    @           0x917df8 testing::TestInfo::Run()
    @           0x917ed5 testing::TestCase::Run()
    @           0x91818f testing::internal::UnitTestImpl::RunAllTests()
    @           0x9184c3 testing::UnitTest::Run()
    @           0x47258b main
    @     0x7f0b398e2840 __libc_start_main
    @           0x47a769 _start
    @                0x0 (unknown)
Makefile:560: recipe for target 'runtest' failed
make: *** [runtest] Segmentation fault (core dumped)

[ RUN      ] RMSPropSolverTest/1.TestSnapshotShare
*** Aborted at 1601950501 (unix time) try "date -d @1601950501" if you are using GNU date ***
PC: @     0x7ffb802fa27b caffe::SyncedMemory::mutable_cpu_data()
*** SIGSEGV (@0x18) received by PID 25993 (TID 0x7ffb87180280) from PID 24; stack trace: ***
    @     0x7ffb81beb390 (unknown)
    @     0x7ffb802fa27b caffe::SyncedMemory::mutable_cpu_data()
    @     0x7ffb80244481 caffe::Blob<>::CopyFrom()
    @           0x7eab65 caffe::GradientBasedSolverTest<>::TestSnapshot()
    @           0x7ec883 caffe::RMSPropSolverTest_TestSnapshotShare_Test<>::TestBody()
    @           0x919d13 testing::internal::HandleExceptionsInMethodIfSupported<>()
    @           0x91368a testing::Test::Run()
    @           0x9137d8 testing::TestInfo::Run()
    @           0x9138b5 testing::TestCase::Run()
    @           0x913b6f testing::internal::UnitTestImpl::RunAllTests()
    @           0x913ea3 testing::UnitTest::Run()
    @           0x4722eb main
    @     0x7ffb7f671840 __libc_start_main
    @           0x47a4a9 _start
    @                0x0 (unknown)
Makefile:560: recipe for target 'runtest' failed
make: *** [runtest] Segmentation fault (core dumped)

[ RUN      ] RMSPropSolverTest/1.TestLeastSquaresUpdateWithEverythingAccum
*** Aborted at 1601950607 (unix time) try "date -d @1601950607" if you are using GNU date ***
PC: @     0x7fc0f880f27b caffe::SyncedMemory::mutable_cpu_data()
*** SIGSEGV (@0x18) received by PID 27660 (TID 0x7fc0ff695280) from PID 24; stack trace: ***
    @     0x7fc0fa100390 (unknown)
    @     0x7fc0f880f27b caffe::SyncedMemory::mutable_cpu_data()
    @     0x7fc0f8759481 caffe::Blob<>::CopyFrom()
    @           0x7e940d caffe::GradientBasedSolverTest<>::CheckAccumulation()
    @           0x9192f3 testing::internal::HandleExceptionsInMethodIfSupported<>()
    @           0x912c6a testing::Test::Run()
    @           0x912db8 testing::TestInfo::Run()
    @           0x912e95 testing::TestCase::Run()
    @           0x91314f testing::internal::UnitTestImpl::RunAllTests()
    @           0x913483 testing::UnitTest::Run()
    @           0x47222b main
    @     0x7fc0f7b86840 __libc_start_main
    @           0x47a3c9 _start
    @                0x0 (unknown)
Makefile:560: recipe for target 'runtest' failed
make: *** [runtest] Segmentation fault (core dumped)
`
the errors focus in these cpp files.
test_gradient_based_solver
test_layer_factory
test_net
test_convolution_layer
test_deconvolution_layer
test_inner_product_layer

also there are some FAILED runtest as follows:

[  PASSED  ] 1997 tests.
[  FAILED  ] 14 tests, listed below:
[  FAILED  ] PoolingLayerTest/0.TestForwardAve, where TypeParam = caffe::CPUDevice<float>
[  FAILED  ] PoolingLayerTest/1.TestForwardAve, where TypeParam = caffe::CPUDevice<double>
[  FAILED  ] CuDNNPoolingLayerTest/0.TestGradientMaxCuDNN, where TypeParam = float
[  FAILED  ] CuDNNPoolingLayerTest/0.TestForwardMaxPaddedCuDNN, where TypeParam = float
[  FAILED  ] CuDNNPoolingLayerTest/0.TestForwardAveCuDNN, where TypeParam = float
[  FAILED  ] CuDNNPoolingLayerTest/1.TestForwardMaxPaddedCuDNN, where TypeParam = double
[  FAILED  ] CuDNNPoolingLayerTest/1.TestForwardAveCuDNN, where TypeParam = double
[  FAILED  ] CuDNNPoolingLayerTest/1.TestGradientMaxCuDNN, where TypeParam = double
[  FAILED  ] LRNLayerTest/0.TestGradientWithinChannel, where TypeParam = caffe::CPUDevice<float>
[  FAILED  ] LRNLayerTest/0.TestForwardWithinChannel, where TypeParam = caffe::CPUDevice<float>
[  FAILED  ] LRNLayerTest/1.TestForwardWithinChannel, where TypeParam = caffe::CPUDevice<double>
[  FAILED  ] LRNLayerTest/1.TestGradientWithinChannel, where TypeParam = caffe::CPUDevice<double>
[  FAILED  ] IOTest.TestDecodeDatumToCVMatContentNative
[  FAILED  ] IOTest.TestDecodeDatumToCVMatContent

vxgu86, 2020-10-06:
by the way, I also tested the BVLC version caffe, it's ok without any errors of FAILED

qianglin-xlnx, 2020-10-09:
@vxgu86 
Please refer to https://github.com/Xilinx/Vitis-AI/issues/164#issuecomment-705927681

vxgu86, 2020-10-10:
thanks, @qianglin-xlnx

closed 2020-10-10 by vxgu86