Xilinx/Vitis-AI #171: How to find dpu interrupt number for device-tree in the .hwh file?

kevinkit, 2020-10-06:
In the device tree one has to define the interrupts, - I was wondering if it is possible to find this information in the hwh file that is generated from vivado?

qianglin-xlnx, 2020-10-09:
@kevinkit 
You can find the prj.xpr file and open it with vivado.
Find the prj.xpr in your project.
find ./ -name "*.xpr"

kevinkit, 2020-10-09:
How does this help me find the interrupt numbers for the device-tree?

qianglin-xlnx, 2020-10-10:
@kevinkit 
After you open the prj.xpr via vivado, you will find how the interrupt signal is connected.  Then, you will find the interrupt number.

kevinkit, 2020-10-10:
Thank you, i got it so far that my Interrupts are connected to the irq1 of the zynq. For the softmax core i use the sixth Input, for the two dpu cores i use the 3rd and the 4th Interrupt. 

Now I know that the interrupt numbers 0-7 are for the irq0, and 8-15 are for irq1. 

Which means that the numbers are 10 and 11 for the dpu cores and 13 for the softmax interrupt. 

However in the dpu user guide the interrupt numbers in the device tree is a three digit number, so what information is encoded in the missing number?

tianfang-fafafa, 2022-02-24:
hi @kevinkit , has this issue been solved? if not, could you please share me with which board you are using and design block with vivado?

closed 2022-02-24 by kevinkit