Xilinx/Vitis-AI #174: Dilation convolution support

Laadr, 2020-10-12:
Hi,

When compiling my network, I get the following error:
[VAI_C][Error] Dilation > 1 is not supported for DepthWise or Deconvolution, current layer is [fc6_separable_conv2d_depthwise].
[VAI_C][Error] Parsing tensorflow model failed.

However, in https://www.xilinx.com/support/documentation/ip_documentation/dpu/v3_2/pg338-dpu.pdf (page 21), it states that dilations are supported in depthwise convolution layers given that the condition "dilation * input_channel ≤ 256 * channel_parallel &&stride_w == 1 && stride_h == 1" is respected.

I precise that I'm using the version of vitis ai available on the main branch last week.

Could you confirm that it is supported? If it is indeed supported, do you have an example of network using such layer in your Vitis AI model zoo?

Thanks in advance for your help

Mookel, 2020-10-15:
Hi, @Laadr 
The dilation > 1 for DepthWise is not supported by our compiler yet.
In thecondition "dilation * input_channel ≤ 256 * channel_parallel &&stride_w == 1 && stride_h == 1" you mentioned, the dilation must be 1.

Laadr, 2020-10-15:
Thanks for your reply. In this case, I hope this feature will be available soon.

closed 2020-10-15 by Laadr

qianglin-xlnx, 2020-10-15:
> 
> 
> Thanks for your reply. In this case, I hope this feature will be available soon.

Yes. It would be available in vai1.3.