Xilinx/Vitis-AI #175: Vitis AI Compiler for Caffe model error #2

JerrySciangula, 2020-10-12:
Hi everyone,

I'm a student at Scuola Superiore Sant'Anna in Pisa.

I'm trying to accelerate a network model on a ZCU102 ultrascale+. 
I can successfully quantize the model using vai_q_caffe and the tool correctly generates the 4 quantize_results files.

When I try to compile the model using vai_c_caffe, it outputs:

[VAI_C][Error] Unrecognized Caffe layer type [Power]
[VAI_C][Error] Parsing layer graph failed

Could anyone explain me the error and how to fix it, if it is possible?

qianglin-xlnx, 2020-10-13:
Hi @JerrySciangula 

TanH Layer [Power] is not supported by the DPU. You can replace it with the other activation function that is supported by the DPU, such as ReLU, RELU6, and Leaky ReLU.
You can find the features of DPU in https://github.com/Xilinx/Vitis-AI/tree/master/DPU-TRD#features or find the details of DPU in PG338 https://www.xilinx.com/support/documentation/ip_documentation/dpu/v3_2/pg338-dpu.pdf

JerrySciangula, 2020-10-13:
Thank you very much!

closed 2020-10-13 by JerrySciangula