Xilinx/Vitis-AI #176: two same Model: ssd-vgg-pruned with different trainval.prototxt and test.prototxt

vxgu86, 2020-10-12:
cf_ssdadas_bdd_360_480_0.95_6.3G_1.2
cf_ssdtraffic_360_480_0.9_11.6G_1.2

the above two models are all with the same model:ssd-vgg-pruned.
but I found that the two trainval.prototxt and test.prototxt of the moels are totally different.
Am I having some misunderstanding?

wangxd-xlnx, 2020-10-13:
Hi，
In fact, these 2 models are different though their structure has some similarities. _cf_ssdadas_bdd_360_480_0.95_6.3G_1.2_ is for the scene of vehicle ADAS inspection, _cf_ssdtraffic_360_480_0.9_11.6G_1.2 ssd_traffic_ is for the scene of security surveillance.
Though the object detection is the same 4 classes, but they are used for completely different field of view scenes.

qianglin-xlnx, 2022-02-17:
hi @vxgu86 

Since we haven't received your reply for a long time, we assume you have solved this issue and I'm going to close it. If you still have any questions, please feel free to reopen it. Thank you very much.

closed 2022-02-17 by qianglin-xlnx