Xilinx/Vitis-AI #177: Hardware design customization for AI application

JerrySciangula, 2020-10-13:
Hi everyone,

I'm a student at Scuola Superiore Sant'Anna in Pisa.
I'm trying to accelerate a network model on a ZCU102 ultrascale+ using Vitis AI.

I'm interested in customizing the hardware design for my specific application. Is there a method or a technique that I can exploit to do so? 

I saw that DPU cores can be combined in parallel. Is there any tutorial about this topic?

Thanks in advance to evreyone who will reply.

kevinkit, 2020-10-21:
You can check the tutorial so to find something you may need: https://github.com/Xilinx/Vitis-AI-Tutorials 
In my experience it was the best way to use the TRD for the DPU and just simple "let it create" by the tcl script to create some Hardware and then make sure that the workflow itself works with given examples. After that you can try to tweak your Hardware design. 

When you add a second DPU it is not automatically guaranteed that your throughput will increase, you will have to write your own software to do so, like giving each even frame number to DPU 0 and each uneven frame number to DPU 1 or everything you can imagine. Some of the tutorials are already using more DPU cores

qianglin-xlnx, 2022-02-20:
Hi @JerrySciangula 
Since we haven't received your reply for a long time, we assume you have solved this issue and I'm going to close it. If you still have any questions, please feel free to reopen it. Thank you very much.

closed 2022-02-20 by qianglin-xlnx